motioneye-project/motioneye #1603: How to setup Google Photos?

jakepals, 2020-01-09:
Thank you, @ccrisan for getting the Google uploads to work again. I have a question on Google Photos, which I cannot find the answer in both here and motioneyeos. How do you setup Google Photos upload? I tried putting the key, and it asks for a location. I tried using a forward slash as a location, but I still receive errors.
What is the correct way to set up Google Photos?

ccrisan, 2020-01-09:
@bob-lee is the author of the Google Photos feature and he's probably the best person to offer help. 

We'll need some more details about the errors you receive and the parameters you use to configure your Google Photos uploader.

bob-lee, 2020-01-09:
@jakepals yap needs more details on parameters and errors on your setting.

jakepals, 2020-01-10:
"Accessing the upload service failed: Bad Request!"

bob-lee, 2020-01-11:
okay, I powered on my raspberry pi again and found gphoto upload is working ok on my side:

1. on chrome browser, confirm I logged in as my gmail account
2. visit motioneye web as admin and see it's working ok
3. change upload service from `Google Drive` to `Google Photo`
4. click `Obtain Key` it shows another tab saying 'motioneye wants to access...' click `Allow` 
5. it presents generated key, copy it by clicking copy icon
6. back to motioneye web, paste obtained key on `Authorization Key` field
7. click `Apply` button on top
8. click `Test Service` should show success message

If you followed here, I guess it should start uploading any captured media to the named (Location field on motioneye) album on your google photo.

@jakepals  Please follow above steps and let me know what you get.
[image]

jakepals, 2020-01-12:
Accessing the upload service failed: object of type 'NoneType' has no len()!

spanishkangaroo, 2020-02-23:
> okay, I powered on my raspberry pi again and found gphoto upload is working ok on my side:
>
> 1. on chrome browser, confirm I logged in as my gmail account
> 2. visit motioneye web as admin and see it's [...]

> Accessing the upload service failed: object of type 'NoneType' has no len()!

I am getting the same errors. Does anyone know how to solve it?

ruinpsilva, 2020-06-17:
@spanishkangaroo and @jakepals, I had the same problem and after I updated the motionEye the only thing I changed was in the location. Instead of "/Kitchen Door" has the example I placed "/files/cameraPhotos. Note that I choose a  folder that didn't existed before.

jakepals, 2020-10-23:
I get this message when testing:
`Accessing the upload service failed: Quota exceeded for quota metric 'All requests' and limit 'All requests per day' of service 'photoslibrary.googleapis.com' for consumer 'project_number:349038943026'.!`

closed 2021-03-05 by starbasessd