motioneye-project/motioneye #1604: usb camera not opened

kd4kw, 2020-01-09:
motioneye says it can not open the usb camera. The camera works great on my RPI 4b with raspian.
I can ssh to the motioneye server (rpi 3b) but can do anything. See example below.
Welcome to meye-c7beefae!
admin@192.168.22.118's password:
[root@meye-c7beefae ~]#
[root@meye-c7beefae ~]#
[root@meye-c7beefae ~]#
[root@meye-c7beefae ~]# ls /etc/motioneye
ls: /etc/motioneye: No such file or directory
[root@meye-c7beefae ~]# cd etc
-sh: cd: etc: No such file or directory
[root@meye-c7beefae ~]# ls
[root@meye-c7beefae ~]# pwd
/root
[root@meye-c7beefae ~]# sudo ls
-sh: sudo: command not found
[root@meye-c7beefae ~]# ls
[root@meye-c7beefae ~]# ls
[root@meye-c7beefae ~]# ls
[root@meye-c7beefae ~]# cd boot
-sh: cd: boot: No such file or directory
[root@meye-c7beefae ~]# dir
-sh: dir: command not found
[root@meye-c7beefae ~]# ls
[root@meye-c7beefae ~]# ?
-sh: ?: command not found
[root@meye-c7beefae ~]# exit

Oh and I can only ssh from my rpi4b not my pc.
any help please.
Thanks.

jasaw, 2020-01-10:
>Welcome to meye-c7beefae!
admin@192.168.22.118's password:

This suggests you are running motionEyeOS on your RPi3b.
MotionEyeOS stores the config files in `/data/etc`, different from Raspbian.
Latest MotionEyeOS has 3 bogus cameras configured by default even though there's an actual camera connected. If you look closely, the 3 bogus cameras are `/dev/video10`, 11, and 12. Actual camera typically starts from `/dev/video0`. @ccrisan Are we able to detect whether `/dev/video10` is an actual camera?
@kd4kw As a workaround, remove all the default cameras, then add a new camera.

As for your ssh problem, it might be a routing issue on your PC. It could be DNS caching issue too. Try clearing your DNS cache. `ipconfig /flushdns` for Windows.

kd4kw, 2020-01-10:
I am able to ssh to the pi. No problem. The issue is the only command it responds to is "pwd" in which it tells me I am in the root directory. No other commands seem to work. I'm sure /dev/video10 is bogus and that's why I can't open the camera. Without terminal access I'm at a loss.
Thanks.

closed 2021-03-05 by starbasessd