motioneye-project/motioneye #1607: Google Photos Upload Stops Working

ukmercanary, 2020-01-11:
Google photos upload not working anymore (Google Drive works ok).
I get this error:

Accessing the upload service failed: Quota exceeded for quota metric 'All requests' and limit 'All requests per day' of service 'photoslibrary.googleapis.com' for consumer 'project_number:349038943026'.!

Cheers

Schwarttzy, 2020-02-01:
I get the same error

aashishvanand, 2020-02-11:
Any fix available? 
Accessing the upload service failed: Quota exceeded for quota metric 'All requests' and limit 'All requests per day' of service 'photoslibrary.googleapis.com' for consumer 'project_number:349038943026'.!

aashishvanand, 2020-02-12:
Suddenly It started working today. Was it due to 10,000 photos upload limit per app per month limitation in google photos?

karlzre, 2020-02-20:
Hi, 

i just set up motioneye on a ubuntu PC, i get the same error "Accessing the upload service failed: Quota exceeded for quota metric 'All requests' and limit 'All requests per day' of service 'photoslibrary.googleapis.com' for consumer 'project_number:349038943026'.!"
I had it yesterday and today.

Any idea what's the problem ?

motionEye Version | 0.42
-- | --
Motion Version | 4.0
OS Version | Ubuntu 18.04

ccrisan, 2020-02-20:
I'm afraid the limit might be set on a project basis and since there's only one project ID for motionEye, that could really cause this problem. Honestly I don't know how we could solve this without asking every user to create a project for themselves in Google Dev. Console and configure it just like I did for motionEye.

Hagai, 2020-03-09:
@ccrisan 
You may request limit increase via the partner program:
https://developers.google.com/photos/partner-program/overview.

karlzre, 2020-03-09:
How can we set our own project ?

ccrisan, 2020-03-09:
> @ccrisan
> You may request limit increase via the partner program:
> https://developers.google.com/photos/partner-program/overview.

I applied to the suggested partner program. Let's see what they have to say.

ccrisan, 2020-03-09:
> How can we set our own project ?

@karlzre there's a pair of Client ID and secret in the source code: https://github.com/ccrisan/motioneye/blob/c955e03798aeea755801c0bf19abcb91a21e24a4/motioneye/uploadservices.py#L150. That points to a motionEye Google project that I've created. You could, in theory change those to your project.

mrlowndes, 2020-04-19:
> @karlzre there's a pair of Client ID and secret in the source code:
>
> https://github.com/ccrisan/motioneye/blob/c955e03798aeea755801c0bf19abcb91a21e24a4/motioneye/uploadservices.py#L150
> . That points [...]

How do I open and modify the source code? Should I just replace the number, or the long string of characters after the dash also?

patstave, 2020-07-09:
I also got this error today. Should I change the service to Google Disk? Is the quota for this service higher?
[image]

patstave, 2020-07-09:
Until the partner program is approved i have changed this to my own project token.
Steps for dummies:

1. Enable the Google Photos API for your Google Cloud account. You probably have to create a project first, but i had one already: https://developers.google.com/photos/library/guides/get-started#enable-the-api

2. Create an OAuth 2.0 client ID: https://developers.google.com/photos/library/guides/get-started#request-id
Select "Desktop" application type.

3. Save the OAuth Client ID and Client Secret for later.

4. Under MotionEyeOS Expert settings (web interface) enable "debug mode" so that file system write access will be allowed. Also make sure SSH is enabled.

5. Connect to to your MotionEyeOS server with SFTP (i used WinSCP for this) and navigate to "/usr/lib/python2.7/site-packages/motioneye" and download the file "uploadservices.py". This is a compiled file, but if you edit this in your favorite file editor the relevant Client ID and secret is actually visible in clear text. The strings you are looking to replace is "349038943026-m16svdadjrqc0c449u4qv71v1m1niu5o.apps.googleusercontent.com" and "jjqbWmICpA0GvbhsJB3okX7s". Take care not to delete the last suffix ASCII character in the compiled file.

6. Upload the modified "uploadservices.py" and replace the old one.

7. Disable debug mode in MotionEyeOS

8. Reboot (maybe not neccessary?)

9. Obtain a new Google Photos authorization key from the MotionEyeOS web interface (File storage -> Upload Service Google Photos) and select your desired directory.

10. That's it!

ukmercanary, 2020-10-08:
I know longer  get the error message however no photos or videos are getting uploaded to Google photos, would anyone care to guess why that might be?  On another note my router had issues and I had to go back to my ISP router...this creates a problem as you can't change the IP address for the profile for the IP camera I have to create a new profile.  Not a massive problem but still a bit of a hassle.     I setup my IP camera using a new profile ..I copied the settings from the old profile..however the video feed has about a 30 second delay thru Motion Eye ( if I view the feed thru the IP cams unbuild browser their is no delay!).   I do like MotionEye but I really just need something that works all the time and is reliable.

jakepals, 2020-10-29:
> Until the partner program is approved i have changed this to my own project token on my local installation.
> Steps for dummies that want to try this on their box:
> 
> 1. Enable the Google Photos API for your Google Cloud account. You probably have to create a project first, but i had one already: https://developers.google.com/photos/library/guides/get-started#enable-the-api
> 2. Create an OAuth 2.0 client ID: https://developers.google.com/photos/library/guides/get-started#request-id
>    Select "Desktop" application type.
> 3. Save the OAuth Client ID and Client Secret for later.
> 4. Under MotionEyeOS Expert settings (web interface) enable "debug mode" so that file system write access will be allowed. Also make sure SSH is enabled.
> 5. Connect to to your MotionEyeOS server with SFTP (i used WinSCP for this) and navigate to "/usr/lib/python2.7/site-packages/motioneye" and download the file "uploadservices.py". This is a compiled file, but if you edit this in your favorite file editor the relevant Client ID and secret is actually visible in clear text. The strings you are looking to replace is "349038943026-m16svdadjrqc0c449u4qv71v1m1niu5o.apps.googleusercontent.com" and "jjqbWmICpA0GvbhsJB3okX7s". Take care not to delete the last suffix ASCII character in the compiled file.
> 6. Upload the modified "uploadservices.py" and replace the old one.
> 7. Disable debug mode in MotionEyeOS
> 8. Reboot (maybe not neccessary?)
> 9. Obtain a new Google Photos authorization key from the MotionEyeOS web interface (File storage -> Upload Service Google Photos) and select your desired directory.
> 10. That's it!

Can this be done in the other motioneye project (not OS version)?

patstave, 2020-10-30:
> Can this be done in the other motioneye project (not OS version)?

I would think so yes.

closed 2021-03-05 by starbasessd