motioneye-project/motioneye #1608: rtsp stream authentication

TeRa2020, 2020-01-12:
Hello ccrisan

First of all : great job

I have an issue trying to connect to my foscam g4P ipcam using motioneye.

The correct url the get the stream is
rtsp://usr:pwd@url:port/directory
e.g.:   rstp://admin:adminspasswd@192.168.178.22:88/videomain

Unfortunately motioneye does  not accept an url like this

Even trying to write the url to netcam_url in the camera-1.conf directly does not work as it seems to be auto corrected and split into netcam:url and netcam:userpass
e.g. 
netcam_url rstp://admin:adminspasswd@192.168.178.22:88/videomain
; netcam_userpass value
becomes 
netcam url rstp://192.168.178.22:88/videomain
netcam_userpass admin:adminspasswd

Using the motion.conf in /etc/motion to set up the ipcam I can get a stream from the ipcam. But unfortunately I cannot manage it using motioneye as motioneye does not handle motion.conf but camera-1.conf and I guess I could have trouble using more than 1 ipcam.

Maybe you could manage motioneye to accept an url like this, please.

Best regards

RaTe

ccrisan, 2020-01-12:
Try using `rtsp://` instead of `rstp://`.

TeRa2020, 2020-01-12:
oh, I am sorry, Just a typing error.
Of course always rtsp:// is in use...

rtsp://admin:adminspasswd@192.168.178.22:88/videomain
is working with VLC for example but not with motioneye.
Behaviour is as described above.

best regards

bakerm99, 2020-01-16:
Hello, 

Firstly i love this project and its helped me out massively thank you.

I had the above problem and i found the solution via another application.

In a nutshell i had to use the following layout

rtsp://192.168.1.18:554/stream0?username=admin&password=????????????

The ip address comes first then the user details. to ensure the connection works use the same program i did https://sourceforge.net/projects/onvifdm/

Hope this helps

digiltd, 2020-01-16:
It still should work, my url with vlc is `rtsp://user:password@192.168.1.123:554/live/ch0` I give motioneye `rtsp://192.168.1.123:554/live/ch0` and the username and password in the relevant fields. Does that not work for you?

closed 2021-03-05 by starbasessd