motioneye-project/motioneye #1609: Dual cameras + Equal resolutions results in no stream

PedroC1999, 2020-01-12:
Hello, 

I bought two 1280x1024 camera sensors off of eBay for cheap and I am using them as cameras for MotionEye on an RPI3B.
Individually they work fine at their rated resolution and framerate, however when paired together, they cannot be set to the same resolution without causing both cameras to fail. Dropping the resolution 1 step lower results in both camera feeds coming back to life. 

They are connected straight to the Pi with no USB hub joining them. 

My goal is to have both cameras running at their highest resolution at the same time

zagrim, 2020-01-12:
RPi3 might not have enough CPU power to be able to decode the feed coming from both of the cameras and again encode it for streaming and/or saving as video. If you don't want to use any lower resolution, you could try with halving the framerate to see if that could work for your purposes. If that doesn't work for you then I'd just suggest getting another RPi for the other camera, or getting something with more CPU capacity instead to do the job.

PedroC1999, 2020-01-12:
Even at 1fps it doesn't work with both 

It appears to be a genuine bug

zagrim, 2020-01-13:
Could you please provide some logs (Motion + MotionEye), tell which versions of Motion and MotionEye you're using, and also check the CPU load of the RPi?

closed 2021-03-05 by starbasessd