motioneye-project/motioneye #1612: Docker installation exec format error

edtwodth, 2020-01-14:
Hi, 

installed as described in documentation

Raspberry pi 3b+

```
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-amd64
```

Get the following error in log:

`standard_init_linux.go:211: exec user process caused "exec format error"`

pvossel, 2020-02-25:
"The image tagged with master-amd64 is built for x86 architecture using the current development branch of motionEye. The image tagged with master-armhf is built for Raspberry Pi architecture."

Use the following: 

docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-armhf

closed 2021-03-05 by starbasessd

lkohari, 2024-05-23:
I have the same problem but on a Raspberry pi.
If I use the armhf version I get errors when I add cameras, and it doesn't work.
```
/usr/bin/ffmpeg: error while loading shared libraries: libudev.so.1: ELF load command alignment not page-aligned
   ERROR: ffmpeg: could find version: Command '/usr/bin/ffmpeg -version' returned non-zero exit status 127
/usr/bin/ffmpeg: error while loading shared libraries: libudev.so.1: ELF load command alignment not page-aligned
   ERROR: ffmpeg: could find version: Command '/usr/bin/ffmpeg -version' returned non-zero exit status 127
/usr/bin/ffmpeg: error while loading shared libraries: libudev.so.1: ELF load command alignment not page-aligned
```
When I use the amd64 version I also get exec format error"
```
exec /bin/sh: exec format error
exec /bin/sh: exec format error
exec /bin/sh: exec format error
exec /bin/sh: exec format error
```

MichaIng, 2024-05-23:
Of course arm64 containers cannot (natively) run on armhf systems. You issue is however an incompatible or broken libudev in the armhf container, it seems.

Did you try the container from [here](https://github.com/motioneye-project/motioneye/pkgs/container/motioneye)?