motioneye-project/motioneye #1613: RPi Zero W - Configuring wired network: no link - panic: rebooting (caused by network)

guarnacciaa, 2020-01-15:
I'm not able to boot image motioneyeos-raspberrypi-20190911.img on a Raspberry Pi Zero W as during the first boot the initialization stops at:

* Configuring wired network: no link
panic: rebooting (caused by network)

wpa_supplicant.conf is in place before the first boot.

Bromley47, 2020-10-19:
I found that despite some advice saying do not put the ssid and password in inverted commas, I did put them in inverted commas (as advised by someone else) and it did work. Maybe worth a try.

closed 2021-03-05 by starbasessd

JMGSE, 2021-05-07:
Same issue but that is not possible to write something. It´s during first startup from new wrote sd-card.
It´s the newest motioneye version. On an other raspberry pi zero (is the same not Wifi) it works fine
Whom can help ?
Thanks
Jörg

starbasessd, 2021-05-07:
I've attached a sample wpa_supplicant.conf file (zipped)
Edit with a Linux-friendly text editor like NotePad++ (DO NOT USE NotePad or WordPad)
Change country to yours
Change SSID to yours (keep quotes)
Change PASSWORD to yours (keep quotes)
Burn correct image to SDCard with Win32DiskImager 
Copy wpa_supplicant.conf to the Boot partition  (the first of the 2 drives showing for the SDCard)
Eject and insert into the Pi2

[wpa_supplicant.zip](https://github.com/ccrisan/motioneye/files/6440787/wpa_supplicant.zip)

starbasessd, 2021-05-07:
I'm sorry, I mis-understood your request for help.
You are using a USB to Ethernet adapter?
Do you have an OTG adapter for the Micro-USB to A-USB? 
Which USB to Ethernet adapter are you using?
Is it supported by RaspberryPiOS without any extra software or drivers?
I use a Belkin USB 2.0 to Ethernet Adapter Model F4U061 with a USB OTG adapter, and it works without any extra software or drivers in both RaspberryPiOS and motionEye 20200606 & dev20201026 (and some older versions) on the PiZeroW V1.1 & PiZero V1.3 (no WiFi).

JMGSE, 2021-05-07:
I use the same on other RPI zeros without any problems but on this I try to configure now it produce this error it´s an ethernet usb adapter on port 2 from the right side one side micro other side A and between the ethernet port. 
configuring wireless network no link
configuring wired network no link
panic: rebooting in260 seconds (caused bey network)
is the error message
the other one I have is UGREEN some has worked fine although

starbasessd, 2021-05-07:
Have you installed RaspberryPiOS on the PiZero to make sure there isn't any problem there?

JMGSE, 2021-05-07:
Nope its MotioneyesOS

JMGSE, 2021-05-07:
I try it would now

JMGSE, 2021-05-07:
It´s a rpi zero W and with the Raspberry OS it works with WIFI

starbasessd, 2021-05-07:
I am seriously confused.

JMGSE, 2021-05-07:
I´m too. I thought that there is a microchip that enables Wifi on the RPI Zero W there are 4 holes nearest to the GPIO and I thought that there is normally the WIFI chip sits but don´t. Now RPI OS is loading updates and before has found My WIFI Points

starbasessd, 2021-05-07:
You should be able to install motionEyeOS on it and copy the wpa_supplicant.conf file to the boot partition (and the ssh.txt file) and have it work normally. 
The best way to know if you have the WiFi version or not is it is labeled on the back
Raspberry Pi Zero V1.3 (no WiFi)
Raspberry Pi Zero W V1.1 (with WiFi)

starbasessd, 2021-05-07:
The WiFi chip is a little TINY silver chip between the ARM chip and the edge with the camera connector.

JMGSE, 2021-05-07:
ok. Thanks very much for today. I would prof it and would let you now what happend ;)
Have a nice weekend. 
Best greetings from Sweden
Jörg

JMGSE, 2021-05-07:
YES it works :D Thanks again. You made My day :D