motioneye-project/motioneye #1614: Is motionEye available in OpenWrt?

probonopd, 2020-01-18:
Is motionEye available via `opkg` in OpenWrt?

kleini, 2020-01-19:
I don't think, that hardware underneath of OpenWrt router software is powerful enough to decode video streams. It is not available and maybe does not even make sense to think about porting it there.

closed 2020-01-19 by kleini

probonopd, 2020-01-19:
OpenWrt can run on hardware such as generic x86_64 and various powerful ARM platforms. So it may well be worth a try. The `motion` daemon runs rather well on it.

kleini, 2020-01-19:
Okay, then it may make sense if the `motion` daemon runs well. But the answer remains the same, that it is currently not available. But motioneye is only Python 2. Get Python 2 running on OpenWrt and then motioneye should not be a problem any more and the guides for other Linux distributions should fit somehow.