motioneye-project/motioneye #1615: Web Interface not loading

fw5k, 2020-01-18:
This is my first time trying out MotionEye and followed the installation instructions exactly for Ubuntu 16.04. So far I'm unable to access the web interface and not sure what the issue is. I have the IP address and port set in the config file and have port forwarding setup too.

I'm not sure what to make of the log info so any help would be very appreciated

`Jan 18 16:33:43 MCserver systemd[1]: Started motionEye Server.
Jan 18 16:33:43 MCserver motion[599]:  * Starting motion detection daemon motion
Jan 18 16:33:43 MCserver motion[599]:    ...done.
Jan 18 16:33:43 MCserver meyectl[593]:     INFO: hello! this is motionEye server 0.42
Jan 18 16:33:44 MCserver motion[673]: [37454208] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motion/motion.conf
Jan 18 16:33:44 MCserver motion[673]: [37454208] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
Jan 18 16:33:44 MCserver motion[673]: [37454208] [ERR] [ALL] myfopen: Error opening file /var/log/motion/motion.log with mode a: Permission denied
Jan 18 16:33:44 MCserver motion[673]: [37454208] [EMG] [ALL] motion_startup: Exit motion, cannot create log file /var/log/motion/motion.log: Permission denied
Jan 18 16:33:44 MCserver meyectl[593]:     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
Jan 18 16:33:44 MCserver meyectl[593]:     INFO: cleanup started
Jan 18 16:33:44 MCserver meyectl[593]:     INFO: wsswitch started
Jan 18 16:33:44 MCserver meyectl[593]:     INFO: tasks started
Jan 18 16:33:44 MCserver meyectl[593]:     INFO: mjpg client garbage collector started
Jan 18 16:33:44 MCserver meyectl[593]: Traceback (most recent call last):
Jan 18 16:33:44 MCserver meyectl[593]:   File "/usr/local/bin/meyectl", line 8, in <module>
Jan 18 16:33:44 MCserver meyectl[593]:     sys.exit(main())
Jan 18 16:33:44 MCserver meyectl[593]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 255, in main
Jan 18 16:33:44 MCserver meyectl[593]:     server.main(arg_parser, sys.argv[2:], command[:-6])
Jan 18 16:33:44 MCserver meyectl[593]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 439, in main
Jan 18 16:33:44 MCserver meyectl[593]:     run()
Jan 18 16:33:44 MCserver meyectl[593]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 395, in run
Jan 18 16:33:44 MCserver meyectl[593]:     application.listen(settings.PORT, settings.LISTEN)
Jan 18 16:33:44 MCserver meyectl[593]:   File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 2042, in listen
Jan 18 16:33:44 MCserver meyectl[593]:     server.listen(port, address)
Jan 18 16:33:44 MCserver meyectl[593]:   File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver.py", line 143, in listen
Jan 18 16:33:44 MCserver meyectl[593]:     sockets = bind_sockets(port, address=address)
Jan 18 16:33:44 MCserver meyectl[593]:   File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.py", line 168, in bind_sockets
Jan 18 16:33:44 MCserver meyectl[593]:     sock.bind(sockaddr)
Jan 18 16:33:44 MCserver meyectl[593]:   File "/usr/lib/python2.7/socket.py", line 228, in meth
Jan 18 16:33:44 MCserver meyectl[593]:     return getattr(self._sock,name)(*args)
Jan 18 16:33:44 MCserver meyectl[593]: socket.error: [Errno 99] Cannot assign requested address
Jan 18 16:33:44 MCserver systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Jan 18 16:33:44 MCserver systemd[1]: motioneye.service: Unit entered failed state.
Jan 18 16:33:44 MCserver systemd[1]: motioneye.service: Failed with result 'exit-code'.`

zagrim, 2020-01-19:
To me it looks like the address you have set in motioneye.conf (`listen`) is such that doesn't match any network interface on the system, or at least for some other the application is not able to bind to it.

fw5k, 2020-01-19:
Ok this is embarrassing... looks everything on motion's end was fine. I forgot to add the default ports for live stream and Web UI into Ubuntu's firewall.

closed 2020-01-20 by fw5k

zagrim, 2020-01-21:
If the problem exists no longer, please close this issue.

zagrim, 2020-01-21:
Great that you could fix it!