motioneye-project/motioneye #1617: Crop/Zoom MMAL

epiller, 2020-01-19:
Is it possible in any way to crop/zoom in an MMAL (Raspberry Pi official) camera?

Pi-DD, 2020-01-28:
Try changing the video resolution in the video device settings.

epiller, 2020-01-29:
Doesn't help. It shows the same picture with lower resolution :(

Pi-DD, 2020-01-29:
Pity! 

I don’t know if there is a way of doing this with MotionEye, but, with RPi-Cam-web-interface it can be done by setting the “sensor region”. I’ve used it to crop down to the part needed only.

https://elinux.org/RPi-Cam-Web-Interface

I assume this is either handled by Motion, which underlies both, or by the Pi Camera software, in which case there may be a way of setting it for or passing it via Motion.

Sorry, I haven’t the time to follow through, but it might give you a toehold to work out and answer.

Good luck!

epiller, 2020-01-29:
Hey, first of all, thanks for the help!
I'm actually using RPi Cam Web Interface right now as it streams so smoothly (way more FPS than motion/MotionEye) and I've managed to setup zooming using the exact same feature you said.
I haven't tried passing those commands to MotionEye thinking it probably won't accept them and since I've already installed the system I'm unable to relocate the camera to a MotionEye-enabled system :(
If someone (reading this) can try, that'd be great!

Pi-DD, 2020-01-29:
Hi, glad it was of help! 

I use RPi-cam as well as MotionEye. I installed the former first (by accident, no plan!) and fortuitously found the sensor region feature because my camera is fronted by a wide angle device not designed for it, so you lose a lot of the possible sensor size.  It works really well and I have it linked to IFTTT for alerts. That works OK because it is indoors - few false positives. 

I experimented with MotionEye for that setup- but had exactly the problem you had - and returned to RPi-Cam because I knew it worked! 

I have just had a “play” passing parameters in MotionEye to Motion and on the the Pi Camera. I didn’t try the complicated string you would need for “sensor region” (lots of trial and error needed to get right values!) but I tried a few others and they work.

FYI, in case you ever try to do it, the format is this.....

mmalcam_control_params -param1 -param2 value(s)-for-2 -param3

As one string up to 4095 characters.

Pi-DD, 2020-01-29:
Sadly geeky afternoon during which I have had time to try this out ....

mmalcam_control_params -roi 0.5,0.5,0.25,0.25

Worked perfectly! For explanation.....

https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

epiller, 2020-02-02:
Dude, thank you so much!
I'm going to apply this solution it in the upcoming days (will need to assemble another RPi system).
And honestly, this should be in the documentation of MotionEye @ccrisan 
Awesome find!

closed 2020-02-02 by epiller