motioneye-project/motioneye #1618: error setting working schedule

mike-sherman, 2020-01-20:
I'm not sure if this issue is within motioneye or elsewhere, but motioneye seems the most likely candidate.

my system is several Raspberry Pi Zeros with cameras, each running motioneyeOS.
These are communicating with a Raspberry Pi (3, now a 4) running HassIO OS and the motioneye add-on.

I am controlling the motion configuration on the Zeros from the motioneye on the HassIO.

I enable a working schedule on each camera manually when I am home to enable motion capture only at night.

Each time I enable working schedule, I have to enable it for each day of the week and reset the start and end times.

After I enter the times for one camera, the same times appear for all of the other cameras, but are RED and can not be saved. Each has to be changed to be accepted. Then they have to be changed back to the desired time.

kurtzilla, 2020-01-29:
Have you tried editing the individual config files for each camera? 
Not offering this as a solution, but as a step to aide in troubleshooting.

mike-sherman, 2020-01-31:
Hi!
I have not tried editing directly to each camera.
I'm "off site" at the moment, but I'll try when I get home.

Thanks,
-Mike

closed 2021-03-05 by starbasessd