motioneye-project/motioneye #1620: CPU usage

speeskeek, 2020-01-24:
[image]
hello all,

I'm very please with motioneye :1st_place_medal: , have it for 4 (?) years now
I have currently 8 camera's running on motioneye on a debian install. 
It is a Xeon E5420 2.5Ghz (quad core). (= an old workstation).
the camera's are 800x600 10-12fps, motion detection is on.
Load of the system is ±1.1.

I tried to change the camera settings to 720p, but then the system chokes.

Is this normal? Or do I have to change something?
Beacause I see peaple here doing Motioneye with a Raspberry on full HD.... :confused: 
I have also a PowerEdge R320 here I can use (Intel Xeon E5-2407 2.20GHz), a bit newer but not faster I guess.

thank you for the input?

gardiol, 2020-02-10:
I have 11 cameras on a i5-4590T CPU @ 2.00GHz, load is around 20-40% all the times, no peaks.
9 cameras are 800x600
1 camera is 1080p

ceeeru, 2020-03-30:
> I have 11 cameras on a i5-4590T CPU @ 2.00GHz, load is around 20-40% all the times, no peaks.
> 9 cameras are 800x600
> 1 camera is 1080p
@gardiol :
Maybe You use hwaccel?

gardiol, 2020-03-30:
Hwaccel? I have no idea, for sure i did not set it up manually... And since i use Gentoo i doubt it's automatically configured... Any way i can check if it is on?

ceeeru, 2020-03-30:
what says intel_gpu_top command?

gardiol, 2020-03-30:
0% on all numbers

ceeeru, 2020-03-30:
I think then Your gpu is not working on encode-decode chain.

closed 2021-03-05 by starbasessd