motioneye-project/motioneye #1622: reolink camera

pftg4, 2020-01-25:
cant install reolink camera get grey screen with unable to open device

rtsp://192.168.1.245:554/h264Preview_01_main

is what i'm using + user and password

error log

   ERROR: mjpg client timed out receiving data for camera 1 on port 8081
   ERROR: mjpg client timed out receiving data for camera 1 on port 8081

thanks for your time

Dave

razvanrudy, 2020-02-02:
Same problem here. I just get a grey/pixelate screen , I can see the stream fps moving up and down but there is no video/image. My camera is using a resolution of 2560x1440 at 30fps. On meye I matched this  settings but no luck. 
Runing motioneyeos-raspberrypi4-20190911

petmi627, 2020-02-03:
HI @pftg4 and @razvanrudy 

Your configuration seems to be right, I used the fallowing `rtsp://<IP>:554/h264Preview_01_main` url and it worked.

But I saw that the reolink camera has two mode one Full HD and one SD. Try it with this one `rtsp://<IP>:554/h264Preview_01_sub`

It can be that the raspberry has not enough cpu to handle the camera, my configuration works but I use motioneye in a docker container on an old computer. 

Another reason would be that the rtsp port is disabled but it is enable per default.

razvanrudy, 2020-02-03:
Fixed the problem. I was using the admin user and password. Created a
common user on reolink wimdows client and used that credentials and now is
working fine.

Wiggins2610, 2020-02-12:
Same problem here, please help me. 
motionEye Version | 0.42
Motion Version | 4.2.2
OS Version | Debian 9.11

Camera: Reolink RLC410 5MP

I use this link in vlc completely working: rtmp://192.168.178.108:1935/bcs/channel0_main.bcs?channel=0&stream=0&user=<user>&password=<pw>

What am I doing wrong?

petmi627, 2020-02-12:
Hi @Wiggins2610,

Do you also set up the username and password while adding the camera, this credentials are the same as the Reolink Camera Webinterface credentials.

The Url seems to be right.

Wiggins2610, 2020-02-12:
Hi @petmi627 
thanks for the quick reply....yes I did. I tried it once again for both links. Either with user and password and either without. 
Same result. Getting frustrated, cause the links are fine. They are working in my browser/vlc-player. 
Any other suggestions? 

rtmp://192.168.xxx.108:1935/bcs/channel0_main.bcs?channel=0&stream=0&user=user&password=pw

rtsp://192.168.xxx.108:554/h264Preview_01_main

Tekno-man, 2020-05-10:
I am having the same issue. RTMP works and RTSP works in HA but in motion eye RTSP just give a grey screen. Has anyone achieved an RTSP with full 5mp resolution? I read somewhere that motion eye doesn't like the unusual video resolution. I have tried both an admin user and a guest user login.

francisuk1989, 2020-07-10:
Since motioneye is based on ffmpeg do you still experience the same issue?

ffmpeg -i rtsp://username:passwordJ@192.168.9.250:554/h264Preview_01_main ipcamrec.mp4

Give this about 30 seconds then hit ctrl + c to stop recording and then playback the ipcamrec.mp4 and see if this gives the same effect.

jacobtomlinson, 2020-08-10:
I had the same problem. My password had a `?` in it and I changed it to something else and it started working.

closed 2021-03-05 by starbasessd

rpwarren, 2021-03-09:
Same.  I removed a "#" from my password and restarted the add-on.  All good.