motioneye-project/motioneye #1623: Motion.conf blank

Saku241, 2020-01-25:
Hello, 
Due to a bug, my 'motion.conf' in 'etc/motioneye/' is blank. 

Can someone send me his file ? Thanks

zagrim, 2020-01-26:
The default contents for one camera are like this:

```
# @enabled on
# @show_advanced on
# @normal_password 
# @admin_username admin
# @admin_password 
# @normal_username user

webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost on

camera camera-1.conf

webcontrol_interface 1
```

Would you care to share what bug you are talking about, so it could perhaps be fixed? Or did you have your own scripting modifying that file?

Saku241, 2020-01-26:
Thank you !

Actually I just added a second camera and after a moment I tried to reconnect and it said Invalid Creditentials so I Check motion.conf and it went blank mysteriously

zagrim, 2020-01-27:
Uh, that sounds like something not easily reproducible. Do you have the logs on that time still available? Was it anyway able to add the second camera?

closed 2021-03-05 by starbasessd