motioneye-project/motioneye #1630: Working schedule starts - but recording won't stop at end time

kurtzilla, 2020-01-29:
I am able to start a recording during the time specified in the working schedule, but the recordings continue to run after the time window has expired. (There have also been some issues getting the recording to start reliably on time - but one issue at a time)

Could this be due to motion being detected near the end time and this continues the recording?

Just trying to record a 15-minute time slot every day and finding that this software may not be appropriate for that purpose. Hoping there is a workaround.

closed 2021-03-05 by starbasessd

kngharv, 2022-06-03:
I am running into the same problem:  working schedule started fine, but the camera is ignoring when working schedule ends.

I am running the latest motioneye via docker:
motionEye Version | 0.43.0
Motion Version | 4.3.2
OS Version | Linux 3.10.105

I've configured the working schedule via GUI, and the camera-3.conf file look like this:

`# @working_schedule 02:00-06:30|02:00-06:30|02:00-06:30|02:00-06:30|02:00-06:30|02:00-06:30|02:00-06:30`
`# @working_schedule_type during`

Would someone give me a pointer on why motioneye ignoring the end of working schedule?  I can't find any clue in motion.log and, when i do a docker logs, i can't see any message on scheduling neither.

thanks

kngharv, 2022-06-05:
Any pointers on how to provide meaningful logs to help developers to debug the problem?