motioneye-project/motioneye #1634: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-B9eBif/pycurl/

seaniedan, 2020-02-04:
pi@pi4:/etc/motion $ pip install motioneye --upgrade
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
  Using cached https://files.pythonhosted.org/packages/48/03/fdfeb93cb6e6540c20910d248095603051162ee1a4738e73b7bb43608777/motioneye-0.42.tar.gz
Requirement already satisfied, skipping upgrade: jinja2 in /usr/lib/python2.7/dist-packages (from motioneye) (2.10)
Requirement already satisfied, skipping upgrade: pillow in /usr/lib/python2.7/dist-packages (from motioneye) (5.4.1)
Collecting pycurl (from motioneye)
  Using cached https://files.pythonhosted.org/packages/ef/05/4b773f74f830a90a326b06f9b24e65506302ab049e825a3c0b60b1a6e26a/pycurl-7.43.0.5.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-B9eBif/pycurl/setup.py", line 961, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-install-B9eBif/pycurl/setup.py", line 623, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-install-B9eBif/pycurl/setup.py", line 101, in __init__
        self.configure()
      File "/tmp/pip-install-B9eBif/pycurl/setup.py", line 238, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-B9eBif/pycurl/

CRImier, 2020-02-04:
Hi! Try to do `apt install python-pycurl` and then run the `pip install` command again. [More info can be found here](https://github.com/ccrisan/motioneye/issues/1562).

CRImier, 2020-02-04:
Also, did you run pip with sudo when first installing motioneye?

seaniedan, 2020-02-04:
Apologies, I hadn't installed motioneye properly! I need to read instructions!

closed 2020-02-04 by seaniedan