motioneye-project/motioneye #1638: Too much upload traffic with motion / motioneye

Oriden, 2020-02-08:
Hello,

Right now I am running Motion + Motioneye on a RPI 3B. There are 4 cameras connected and continuously recording on there. My upload speed on the rpi is around 0.

Now I'm looking to migrate this to my server with Docker containers, I'm using Docker Compose for this. I used the motion & motioneye docker images and set up the infrastructure & cameras, but for some reason, the upload speed is around 11 MB/s. Ofcourse this is all internal so it doesnt matter much, but it slows my SSH server alot and thus, very annoying. 

Any idea why this could be? This is the only place I could find for support.

Thanks!

zagrim, 2020-02-10:
Is there something missing from this: "My upload speed on the rpi is around 0."? I mean, you have 4 cams recording, so the data amount uploaded should match the amount of generated video stream, and that can't be around zero unless you have very low frame rate and resolution?

Regardless of the actual difference, one thing you should check is if you changed some of the output video settings when migrating on Docker. Resolution, frame rate and encoding ("format") all affect the bitrate of the video.

martinsirbe, 2020-04-04:
I use 1 RPi camera with 720p resolution set in motioneye config, when I open motioneye on port `8765`, I see a single camera feed. Observing how much data is sent from RPi to my browser, to my surprise it was whooping 100 MB per minute (more sunny day even 200 MB per minute). Opening dev tools, I see that around 180 HTTP GET requests are made to retrieve JPEG. Each request is around 600KB. While using netcat with raspivid output to stream the camera feed directly to my laptop, 1 minute only 0.720 MB was sent!! Same resolution, almost 0 latency. However, as motioneye makes HTTP GET request to retrieve each frame results in choppy video feed, with a massive latency. Can someone explain this?

closed 2021-03-05 by starbasessd