motioneye-project/motioneye #1639: How do you solve the "Notification email failed: 5.7.9" error for Motion Notifications?

ygreq, 2020-02-10:
Hi there! 

I get this message whenever I try to test the setup I made in Motion Notifications. 

> Notification email failed: 5.7.9 Application-specific password required. Learn more at 5.7.9 https://support.google.com/mail/?p=InvalidSecondFactor z19sm307897wmi.43 - gsmtp

I cannot find a solution online. Any ideas how to get past this? 
Thank you so much
ygr

Heisenberg389, 2020-03-12:
One of the reason could be that your 2 step verification for google is activated, in that case, the email notification won't work, try turning it off or try using a different email.

gozmon42, 2020-05-04:
you need an "app password" for 2-factor authentication in gmail.
https://support.google.com/mail/answer/185833?hl=en
I had to tell google it was an OSX system, but it still worked fine.

closed 2021-03-05 by starbasessd