motioneye-project/motioneye #1640: Python3 support?

gardiol, 2020-02-10:
As you know, Python 2 has been discontinued and support has stopped by the end of Jannuary 2020. Most distributions are going to fade Python 2 out in the near future. 

Is MotionEye going to be ported to Python3?

zagrim, 2020-02-10:
It's in progress, see #1208 and #1572

Aspire1Inspire2, 2020-02-28:
[https://lists.ubuntu.com/archives/ubuntu-devel/2020-February/040918.html](url)
Sorry to interrupt here, I know the work is in progress and we are moving to Python 3.

As mentioned in the official Ubuntu conmmunication, the python 2 kernel will be removed from Ubuntu in the future. However, 

> 20.04 LTS will ship python2. Derivatives who cannot port some applications are able to use python2 for the 20.04 LTS release.

Does that mean motioneye will be directly functional in 20.04 LTS even if I update Ubuntu without bumping up motioneye to Python 3?

gusarg81, 2020-05-07:
@Aspire1Inspire2 well, is not functional for me. It does not work after upgrading from 18.04 to 20.04. Besides there is no way (from the base python2 packages from the os) to reinstall motioneye since they removed pip package.

In my case for know I've installed Zoneminder (but I dont like at all... UI interface for me has a so bad implementation, full of multiple browser Windows instead of JS dialogs, etc), and it freeze a lot (which also trying to just restart zoneminder takes a lot).

We just have to wait for Python3 implementation of Motioneye.

Prutsium, 2020-05-10:
Same here .... following this one

Prutsium, 2020-05-12:
Seems there is an Python3 version available and got it running on my Ubuntu 20.04 system now.

gusarg81, 2020-05-12:
> Seems there is an Python3 version available and got it running on my Ubuntu 20.04 system now.

In my case, installed latest git from Python3 branch, gives me:

> may 12 11:14:14 home meyectl[320988]:   File "/usr/local/lib/python3.8/dist-packages/motioneye-0.42-py3.8.egg/motioneye/utils.py", line 964, in parse_editable_mask_file
> may 12 11:14:14 home meyectl[320988]:     for y in range(ny):
> may 12 11:14:14 home meyectl[320988]: TypeError: 'float' object cannot be interpreted as an integer
> may 12 11:14:14 home meyectl[320988]:    ERROR: 500 GET /config/list/?_=1589292854791&_username=admin&_signature=cd653bde59503c14c8a6c78f3324cdfd3c2ddbc8 (10.0.1.1) 101.23ms

gusarg81, 2020-05-12:
Also, deleted all existen conf files (cameras) and started from Scratch. I can't also add cameras, when I add the URL for Remote cameras:

> may 12 11:27:33 home meyectl[323621]:   File "/usr/local/lib/python3.8/dist-packages/motioneye-0.42-py3.8.egg/motioneye/utils.py", line 535, in on_server
> may 12 11:27:33 home meyectl[323621]:     identifier = re.findall('Server: (.*)', data)[0].strip()
> may 12 11:27:33 home meyectl[323621]:   File "/usr/lib/python3.8/re.py", line 239, in findall
> may 12 11:27:33 home meyectl[323621]:     return _compile(pattern, flags).findall(string)
> may 12 11:27:33 home meyectl[323621]: TypeError: cannot use a string pattern on a bytes-like object

Aspire1Inspire2, 2020-05-15:
As more and more people upgrade to Ubuntu 20 LTS, it will break existing python2 installation of motioneye as found by comments in this issue.
Could anyone give a hint on what the upgrade path will look like in the future?

mikeonbike, 2020-07-02:
I'm also looking for guidance on using the Python3 version.  It looks like that branch is both ahead/behind dev.  No changes in 4 months?  I'm a bit out from getting my cameras installed so wouldn't mind beta testing.

I will pull the Python3 branch and see how it installs/runs on Mint/Ubuntu 20.

Signum, 2021-03-05:
So it's working with Python3 now?

starbasessd, 2021-03-05:
There are updated instructions to work around the EOL of Python2.7 and pip2 until motionEye/motionEyeOS are updated.

nullr0ute, 2021-03-05:
Why is this being closed then? It should really be left open until proper python3 support lands rather than work arounds.

starbasessd, 2021-03-05:
THere are other issue tickets and PRs for the Python 3 upgrade. Closed as
duplicate.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jjpd19, 2022-01-23:
I have just installed motioneye following the given instructions on a raspberry pi zero (completely updated), and I am having the issue:
ene 23 07:54:32 raspberrypi systemd[1]: Started motionEye Server.
ene 23 07:54:32 raspberrypi meyectl[2310]: Traceback (most recent call last):
ene 23 07:54:32 raspberrypi meyectl[2310]:   File "/usr/local/bin/meyectl", lin>
ene 23 07:54:32 raspberrypi meyectl[2310]:     from motioneye.meyectl import ma>
ene 23 07:54:32 raspberrypi meyectl[2310]:   File "/usr/local/lib/python3.9/dis>
ene 23 07:54:32 raspberrypi meyectl[2310]:     import settings
ene 23 07:54:32 raspberrypi meyectl[2310]: ModuleNotFoundError: No module named>
ene 23 07:54:32 raspberrypi systemd[1]: motioneye.service: Main process exited,>
ene 23 07:54:32 raspberrypi systemd[1]: motioneye.service: Failed with result '>
~

gardiol, 2025-02-04:
You really bumping a 3 years old critics ticket?

feurig, 2025-02-04:
Sorry. I can't get motioneye to install on debian bookwork or trixie. And it all comes down to still not supporting python3 in one way or another. Just ignore me. I am over it.

zagrim, 2025-02-04:
@feurig (and anyone coming here later on) you'd perhaps be more likely to get help by creating a new issue explaining what goes wrong after you've followed the installation instructions here: https://github.com/motioneye-project/motioneye?tab=readme-ov-file#installation