motioneye-project/motioneye #1641: Latest Buster updates on Raspberry Pi appear to break Motioneye

Pi-DD, 2020-02-13:
Applied the latest Raspbian system updates last night. 

As from this morning captured MP4 video will not play. 

The error message displayed on iOS is..

Error: Media format unsupported or otherwise unvilable/unsuitable (sic) for playing

Using VLC player on Raspbian and Windows 10 they just play as a black screen

Files recorded the previous day play OK in all environments.

If I stop Motioneye and run raspivid it records an MP4 just fine, boxing it with GPAC

Pi-DD, 2020-02-13:
I've now done my homework a bit more and see that they syslog message is probably not relevant, as it appears in other devices that are recording just fine, both Stretch and Buster. 

Now puzzled about why this device alone seems to have the problem when others do not.

wx-watch, 2020-02-13:
I’m curious if you are having a similar issue as I am, where the MPEG profile is causing issues playing on iOS/macOS. 
https://github.com/ccrisan/motioneye/issues/1611

Pi-DD, 2020-02-13:
Not sure I fully understand your question. What profile do you mean?

All I can say is that this is the first time this has happened. Up to know files have played OK on iOS and Win 10 (I don’t have a Mac).  And (I have now found out.) is only happening with files created on this one device since yesterday. I have another Pi Zero   at the same level and config that works just fine.

wx-watch, 2020-02-13:
Do you happen to have ffprobe installed? If you run "ffprobe file.mp4" it should return a line that contains the mpeg4 profile, such as:

Stream #0:0(und): Video: h264 (High 4:4:4 Predictive)

I think the High 4:4:4 Predictive is problematic for iOS and macOS. If I do the passthrough mode for my RTSP camera, there's no post-processing done (to overlay the timestamp) and the profile instead looks like this:

Stream #0:0(und): Video: h264 (Main) 

...and it plays fine for me. But I can't seem to find a way to control the mpeg4 profile itself, and was wondering what ffprobe shows for you on one of the problem videos vs the older ones. Thanks!

Pi-DD, 2020-02-13:
Hi - as it happens ffprobe is installed.

I ran it for a file from today and one from yesterday - from the same pi- today’s file does not play, yesterday’s does. The ffprobe results are...

Yesterday’s file.....
Stream #0:0(und): Video: h264 (Constrained Baseline) (avc1 / 0x31637661), yuv420p, 640x480, 919 kb/s, 1.80 fps, 5 tbr, 10240 tbn, 10 tbc (default)

Today’s file....
Stream #0:0(und): Video: h264 (High 4:4:4 Intra) (avc1 / 0x31637661), yuv420p, 640x480, 4863 kb/s, 3.97 fps, 19.92 tbr, 10240 tbn, 10 tbc (default)

The only “change” between the two is that I ran Raspbian updates.

If I run ffprobe for a file created on my other Zero, today, it gives....

Stream #0:0(und): Video: h264 (Constrained Baseline) (avc1 / 0x31637661), yuv420p, 640x480, 847 kb/s, 2.02 fps, 14.83 tbr, 10240 tbn, 10 tbc (default)

But that had Raspbian updates applied last night too! 

And if I run ffprobe on files it created yesterday I get.......

Stream #0:0(und): Video: h264 (Constrained Baseline) (avc1 / 0x31637661), yuv420p, 640x480, 854 kb/s, 2.14 fps, 5 tbr, 10240 tbn, 10 tbc (default)

It seems “High 4:4:4” is problematic - but I have no clue how it has suddenly become the optioned used in one Pi!

Pi-DD, 2020-02-13:
I just checked in real detail the MotionEye options from the web interface. The only difference I could find between the two cameras was that, somehow, under “video streaming” (for the one producing unplayable files) “streaming image resizing” was turned on. 

Don’t recall doing that, but I have set it to “off” and let’s see what happens tomorrow! Bit of a stab in the dark, but I observe that even when video streaming is “off” it seems to affect what is happening.

Pi-DD, 2020-02-14:
Bingo! I can only conclude that changing the “streaming image re-sizing” did the trick since I now have playable videos and ffprobe gives this...

Stream #0:0(und): Video: h264 (Constrained Baseline) (avc1 / 0x31637661), yuv420p, 640x480, 1174 kb/s, 1.87 fps, 5 tbr, 10240 tbn, 10 tbc (default)

This is all way beyond my technical skill/knowledge levels, but perhaps somebody suitably knowledgeable can get to the bottom of it.

I’m not going to close this yet to give @wx-watch time to see it and respond.

wx-watch, 2020-02-14:
Hmm - my Streaming image re-sizing is already set to off but let me experiment today and see.

Pi-DD, 2020-02-14:
Good luck with your investigations! 

Maybe one difference is that I’m using the Pi camera and so raspivid is being invoked by the underlying Motion software to capture the video. Is an external source handled differently?

wx-watch, 2020-02-14:
Unfortunately this didn't fix it for me, unless I use passthrough mode the resulting video is still High 4:4:4 Predictive and unplayable on macOS/iOS. I will keep my https://github.com/ccrisan/motioneye/issues/1611 open which asks what exactly controls the mpeg4 profile. 

Glad you are up and running!

Pi-DD, 2020-02-14:
I suspect that the answer lies somewhere in the interface between Motioneye and Motion. The former is “just” a web interface to the latter and it is Motion that actually does the file creation. 

I don’t have anything like enough understanding of either to probe further. It might be that different setting combinations in Motioneye unwittingly direct Motion quite differently. 

You could try posting something in the Github issues for Motion cross referring back to your issue here.

Pi-DD, 2020-02-14:
Having read some stuff about the h264 codec here....

http://blog.mediacoderhq.com/h264-profiles-and-levels/

I’m wondering if, in MotionEye web interface “Movies” section, and in the “Video Streaming” section, the “Movie Quality” parameter drives the one of the same name in Motion 

https://motion-project.github.io/motion_config.html#Configuration_OptionsAlpha

and that, in turn affects the H264 profile used? 

It seems to default to 75%, but you could check your setting and try a lower one to see whether that makes a difference? I did notice that in my settings, in the camera that had a “problem”, it was>75.

wx-watch, 2020-02-14:
OMG! I can't believe I never thought to change the quality slider. I had bumped it to 100% (I figured I had plenty of storage, so why not?) Changed it to 90% and now it also says Video: h264 (Constrained Baseline). And it plays fine!

Two issues closed, thanks to you!

Pi-DD, 2020-02-15:
Glad it worked for you! My pleasure! I hate not understanding why things happen. 

I still haven’t found a really clear, simple, explanation of how ffmpeg “decides” what profile to use,  nor how Motion passes it instruction. But this is often the case in deeply specialised areas - there is much tacitly assumed knowledge that the non-specialist just does not have. 

I’m going to write up a brief summary and then close this issue.

Pi-DD, 2020-02-15:
Summary of problem and fix......

- under the covers of MotionEye, Motion can, in certain circumstances, generate mp4 files that are “unplayable”.

- this is because they have been created with an H264 profile that the playing software will not handle. (Typically h264 (High xxxx). 

- This appears to be because of one or both of two possible factors set in either, or both, of “Movies” and “Streaming Video” settings - Movie Quality and Streaming image Re-Sizing.

- It appears that in particular a high setting for Movie Quality causes this.

- 

-

Pi-DD, 2020-02-15:
Resolved, see above

closed 2020-02-15 by Pi-DD