motioneye-project/motioneye #1642: After selecting "Fast Network Camera" to the ON position: get gray pixellated image only.

bytetick, 2020-02-14:
Issue: Gray pixelled screen only occurs when I toggle on the [ Fast Network Camera ] to the 'on' position, and then when I select "Apply" btn,  the camera's monitor window pane at url: localhost:port is 'grayed out'. The camera (RPi camera on RPI3b) is functioning perfectly: there is a timestamp and all.  If I reverse the aforementioned, and uncheck the "on" toggle to turn it off, and click "Apply" button, the camera functions fine and shows frame outputs as expected. I have read and researched a lot of FAQ and googled a lot. 

I am using the current stable version. Flashed from site. 
Everything works perfectly but when  trying to turn on Fast Network camera to allow for motion detection settings to execute.

jasaw, 2020-02-16:
Can you please try the latest prerelease?

bytetick, 2020-02-16:
Sure. TY.