motioneye-project/motioneye #1643: Limited options on Ubuntu Server

Rickymundo, 2020-02-15:
Hi,

I installed MotioEyeOs on a Pi and that works but I would like to have it running on a barebone server. I installed Ubuntu server and follow your guide. It works now but I have limited options. See screenshot:

[image]

otte845, 2020-03-14:
You can use the config file ( https://github.com/ccrisan/motioneye/wiki/Configuration-File ) to enable Shutdown and reboot control (option enable_reboot), wifi settings edit (option wpa_supplicant_conf) and others, but motioneye have to run as root for them to work

JamBalaya56562, 2026-11-08:
Closing as answered above. The shutdown and reboot buttons appear with `enable_reboot true` in `motioneye.conf`, and the user motionEye runs as, `motion` since 0.43.1, must be allowed to reboot. There is no update button any more: upgrade with pip as shown in the [README](https://github.com/motioneye-project/motioneye#upgrade).

closed 2026-11-08 by JamBalaya56562