motioneye-project/motioneye #1644: Stretch => Buster : motion KO

MacAre, 2020-02-16:
Hi,
I updated my Raspberry from Stretch to Buster with this procedure: http://baddotrobot.com/blog/2019/08/29/upgrade-raspian-stretch-to-buster/

Since the update, motion no longer starts. After the command "sudo systemctl enable motioneye", I have the message:
Synchronizing state of motioneye.service with SysV service script with / lib / systemd / systemd-sysv-install.
Executing: / lib / systemd / systemd-sysv-install enable motioneye
update-rc.d: error: motioneye Default-Start contains no runlevels, aborting.

I uninstalled motion (remove -purge) and deleted /var/lib/motioneye, /etc/motioneye
Then reboot, then I relaunched the installation procedure https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
I still get the error message :-(

Do you have an idea of the problem ?

zagrim, 2020-02-16:
You could try to add some boilerplate comments that sysv services have to motioneye.service file, as instructed here: https://raspberrypi.stackexchange.com/a/89768/37765
You can use the same block of lines given in the answer, just edit the text after `Provides:` and `Short-Description:` to your liking.

My ME Raspbian installation is Stretch, but the file to edit still might be the same in your case: `/etc/systemd/system/motioneye.service`. It's probably anyway good idea to backup the file before modifications.

MacAre, 2020-02-16:
Here is my file /etc/systemd/system/motioneye.service:

```
[Unit]
Description = motionEye Server

[Service]
ExecStart = / usr / local / bin / meyectl startserver -c /etc/motioneye/motioneye.conf
Restart = on-abort

[Install]
WantedBy = multi-user.target

### BEGIN INIT INFO
Provides: sendir
# Required-Start: $ all
# Required-Stop:
# Default-Start: 2 3 4 5
# Default-Stop:
Short-Description: Execute sendir at service termination ...
### END INIT INFO
```

systemctl daemon-reload
systemctl enable motioneye
Error message is still present.

I will configure https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User
Then I will test sudo -u youruser meyectl startserver -b -c /home/youruser/motioneye/motioneye.conf

MacAre, 2020-02-16:
Result of the command : sudo -u pi meyectl startserver -b -c /home/pi/motioneye/motioneye.conf

```
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python2.7/site-packages/motioneye/meyectl.py", line 254, in main
    import server
  File "/home/pi/.local/lib/python2.7/site-packages/motioneye/server.py", line 31, in <module>
    import handlers
  File "/home/pi/.local/lib/python2.7/site-packages/motioneye/handlers.py", line 30, in <module>
    import config
  File "/home/pi/.local/lib/python2.7/site-packages/motioneye/config.py", line 32, in <module>
    import motionctl
  File "/home/pi/.local/lib/python2.7/site-packages/motioneye/motionctl.py", line 28, in <module>
    import mediafiles
  File "/home/pi/.local/lib/python2.7/site-packages/motioneye/mediafiles.py", line 41, in <module>
    import uploadservices
  File "/home/pi/.local/lib/python2.7/site-packages/motioneye/uploadservices.py", line 27, in <module>
    import pycurl
ImportError: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /home/pi/.local/lib/python2.7/site-packages/pycurl.so)
```

zagrim, 2020-02-16:
The error you mention on ME startup is at least fatal and requires a fix regardless of the systemd service issue. Please take a look at the proposed solutions in #1494 and the linked issues there.

MacAre, 2020-02-16:
Problem solved by uninstalling pycurl and reinstalling it.

```
pip uninstall pycurl
pip install pycurl
```

https://github.com/ccrisan/motioneye/issues/1494#issuecomment-545744664

Thank you zagrim.

closed 2020-02-16 by MacAre