motioneye-project/motioneye #1645: Cannot open video device or no video device available

donatom805, 2020-02-18:
I have used motioneyeOS before and had no trouble getting it up and running on Raspberry Pi 3B.

Recently I purchased a RP 4B and hooked up the same compatible camera that I used on the RP 3B.
I am now using Raspbian Buster full version with motioneye app. I have activated video using raspi-conf, have added the bcm2835-v4l2 driver to /etc/modules and increased RAM to GPU to 256 megabytes.

I can use the attached camera to produce still shots, so I know that the camera works, but whenever I start up motioneye on a browser, I get an error stating that "cannot open video device".

I followed the wiki to install motioneye and everything went smoothly.

Here is the output from journalctl:

[motioneye.log](https://github.com/ccrisan/motioneye/files/4217634/motioneye.log)

Heisenberg389, 2020-03-12:
[image]
@donatom805 if you have a USB camera, try selecting a UVC camera instead of bcm2835-codec-decode.

donatom805, 2020-03-13:
Thanks, @Heisenburg389. I got it to work. I made a very basic mistake: I
neglected to select the proper (attached) camera. Once I did that, it
worked well. I couldn't find my thread, so I wasn't able to note this on
the thread.

closed 2021-03-05 by starbasessd