motioneye-project/motioneye #1646: Remote motioneEye Camera - authentication failed

k-redas, 2020-02-18:
Hi,

I would like to install motioneye sytem with multiple Raspberry pi (zéro, rpi3). I installed motioneye in each of them.
One of them is the "server" where i would like to agregate all camera.
And when i would like to add a remote motionEye camera on my Raspberry, i have a message witch said : "authentication failed".
[image]

Have you some idea about this issue ?
Thank you

k-redas, 2020-03-04:
The problem is solved,

In fact, this device (cam) dont support 5Ghz wifi, but only 2.4Ghz. So you must use the 2.4Gz network wifi at home and you must configure the wpa_supplicant.conf like this :

country=FR
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="xxxxx"
    psk="xxxxx"
    key_mgmt=WPA-PSK
}

jakepals, 2020-03-09:
I'm experiencing the same issue. However, I'm not on wifi. My machine is connected via ethernet.

closed 2021-03-05 by starbasessd