motioneye-project/motioneye #1647: snapshot file name doesn't match video filename (motion detection)

dolhop, 2020-02-18:
It would seem that when using snapshot in motion detection mode, the filename (which is a timestamp) does not match that of the corresponding motion detected video filename - they seem to be off by many seconds.

The video filename seems to correspond to the start of the motion event, but the snapshot doesn't have any correlation to the start or the end event.  I thought it might be because I had configured it to store 10 frames prior to the motion event, but evenafter changing that setting to zero, the issue remains.

My use case is that I would like to run a command on modet that can notify (via email or whatever I choose) by sending the snapshot and a URL to the associated video file.  I realize that there are some potential conflicts there because the video filename isn't known until after the file is stored (though the fact that the filename matches the start of the event tells me that the filename could be made available at the start of the modet event).

zagrim, 2020-02-19:
I've had the impression, that the snapshot would be from such a time that the amount of detected motion is high (highest?) - the start of the video might not really tell what really triggered the recording.

If the command (well, likely a script run as a command) would parse the timestamp of the video and then compare that to parsed timestamps of the snapshots, it could select the snapshot that's closest to the video start time. However, that takes some programming, especially if written as a shell script.

Another workaround might be having `%v` ("event") in the file names of both the snapshot and the video. I am not really sure, though, how Motion numbers the events and if that really would work, but that is likely the only possible thing available in the file name scheme that should match things nicely. If the timestamps are still something you need to have available, with `%v` it only would take parsing of the event identifier out of the video file (you could e.g. surround it with some known characters for easier processing) and pick the matching snapshot file (or vice versa). Still a bit more than a neat one-liner but much easier than parsing and nearest-matching the time stamps.

lucianaohira, 2020-12-02:
Hey, I´ve just used the %v script, and the file name is 01.jpg, and them should be 02.jpg and so on...

closed 2021-03-05 by starbasessd