motioneye-project/motioneye #1648: upgrade on raspbian stretch - problem

angusmedia, 2020-02-18:
pip install motioneye --upgrade
Collecting motioneye
  Using cached https://files.pythonhosted.org/packages/48/03/fdfeb93cb6e6540c209                                                                                  10d248095603051162ee1a4738e73b7bb43608777/motioneye-0.42.tar.gz
Collecting jinja2 (from motioneye)
  Using cached https://files.pythonhosted.org/packages/27/24/4f35961e5c669e96f65                                                                                  59760042a55b9bcfcdb82b9bdb3c8753dbe042e35/Jinja2-2.11.1-py2.py3-none-any.whl
Collecting pillow (from motioneye)
  Downloading https://files.pythonhosted.org/packages/b3/d0/a20d8440b71adfbf1334                                                                                  52d4f6e0fe80de2df7c2578c9b498fb812083383/Pillow-6.2.2.tar.gz (37.8MB)
    99% |████████████████████████████████| 37.8MB 3.5MB/s eta 0:00:01Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in                                                                                   run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepa                                                                                  re_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 620, in _prep                                                                                  are_file
    session=self.session, hashes=hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 821, in unpack_u                                                                                  rl
    hashes=hashes
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 659, in unpack_h                                                                                  ttp_url
    hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 882, in _downloa                                                                                  d_http_url
    _download_url(resp, link, content_file, hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 603, in _downloa                                                                                  d_url
    hashes.check_against_chunks(downloaded_chunks)
  File "/usr/lib/python2.7/dist-packages/pip/utils/hashes.py", line 46, in check                                                                                  _against_chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 571, in written_                                                                                  chunks
    for chunk in chunks:
  File "/usr/lib/python2.7/dist-packages/pip/utils/ui.py", line 139, in iter
    for x in it:
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 560, in resp_rea                                                                                  d
    decode_content=False):
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/res                                                                                  ponse.py", line 432, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/res                                                                                  ponse.py", line 380, in read
    data = self._fp.read(amt)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachec                                                                                  ontrol/filewrapper.py", line 63, in read
    self._close()
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachec                                                                                  ontrol/filewrapper.py", line 50, in _close
    self.__callback(self.__buf.getvalue())
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachec                                                                                  ontrol/controller.py", line 275, in cache_response
    self.serializer.dumps(request, response, body=body),
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachec                                                                                  ontrol/serialize.py", line 87, in dumps
    ).encode("utf8"),
MemoryError

zagrim, 2020-02-19:
Please see this issue for potential solution: #1230

angusmedia, 2020-02-19:
That worked
pip install motioneye --upgrade --no-cache-dir
systemctl restart motioneye

angusmedia, 2020-02-19:
...but all cams disapeared - had to configure them again

Joshua2504, 2020-02-20:
> ...but all cams disapeared - had to configure them again

Yeah that's why you don't use motioneye. I would really recommend you to use Shinobi. it won't eat all your CPU plus you can just export your configs. Aaaaand it comes with a PROPER MADE interface.

don't use motioneye.

angusmedia, 2020-02-20:
Import and Export is possible in motioneye.

zagrim, 2020-02-20:
The cams disappearing might be due to missing the instructions for config migration when upgrading from 0.3x to 0.40 (which are given in the [release notes](https://github.com/ccrisan/motioneye/releases/tag/0.40) which might not be a place many users go read). In that (or any similar case) the config files are still present, but either they are rejected due to parsing errors (which is shown in the logs) or `motioneye.conf` just lacks the "linking" to individual camera configs. So configuration is not lost, it just takes some quirks to get it back into use.

closed 2021-03-05 by starbasessd