motioneye-project/motioneye #1652: Warning

Joshua2504, 2020-02-20:
Just a quick warning for everyone who's considering to use this:

- CPU eating
- starts itself all the time (HOW EVEN???)
- just a stupid software based on motion. don't use it for your raspberry pi, you will roast it

regards.

PS: SHAME on you for even developing this. LOL.

closed 2020-02-20 by Joshua2504

ccrisan, 2020-02-20:
@Joshua2504 you might want to post this on the Facebook page and the user group as well:
 * https://www.facebook.com/motioneyeos
 * https://groups.google.com/forum/#!forum/motioneye

Otherwise your message won't have the reach you expect.

kleini, 2020-02-20:
- I am properly checking for motion on two 4k streams and recording works as expected in 4k and 25fps on some very old Intel Core i7
- process runs fine in Docker container, no restarting, uptime some hundred days (not more due to updating motioneye and motion)
- yes, motion is the operating software behind motioneye. what exactly is the problem to re-use existing good working software?

zagrim, 2020-02-20:
@Joshua2504 I came to MotionEye **exactly** due to having too high CPU usage with Shinobi on my RPi3, and even after getting the CPU usage to tolerable level the camera streams still were lagging behind some 30+ secs. With Motion and MotionEye things are as fine as they could probably be given that RPi isn't exactly that powerful device. So it's not as simple as you seem to think, even though I might just have chosen gear that produces stream with which Motion is good at (and Shinobi likewise bad).
And IMO the simplicity of camera config in MotionEye is way user-friendlier than the wall-of-options that Shinobi presents. But YMMV of course.

ascillato, 2020-02-20:
I use motioneye on a rpi3 (40% cpu) as docker in home assistant with 3 hd cameras (720p) at 5fps with motion detection and publishing it to mqtt, with continuous stream to 2 wall tablets, with saving to an HDD, with saving on Google drive. Zero restarts nor hungs. So, for me this software rocks and works amazing.

Big congratulations to everyone involved in motioneye. Super happy with it. 

I have just moved to a rpi4 and it uses about 20% cpu. 

@Joshua2504 seems that you are having just a config issue. Don't let your intolerance take you away from using this amazing tool.