motioneye-project/motioneye #1655: No picture in motioneye

zslukacsik, 2020-02-25:
Hi,

Everything working properly but no  picture in motioneye app after connected to server.
My solution: Disabling Chrome app. But after a day the motioneye not working again.
If I enable the Chrome app and manage-delete all data and disable it, the motioneye will work for one day again.
Could you give me a better solution?

Motioneye version: 0.9.7.5.2
Android version: 7.1.1

zagrim, 2020-02-25:
Are you talking about MotionEye Android application? If yes, this is wrong project (this is the application that the Android app builds on (well, I don't know how it actually works).
Maybe this is what you were looking for: https://github.com/JairajJangle/motionEye_app_HomeSurveillanceSystem ?

zslukacsik, 2020-02-25:
Ohh, sorry. THX for your help!

closed 2020-02-25 by zslukacsik