motioneye-project/motioneye #1657: Feature request: Delete button in play mode

kbfifi, 2020-03-01:
Hi

Great project!! I hope this is the place to do requests too. To get in "play mode": click camera view->play button->click recorded video. Now that you have the option to play, pause, close and download, it would really be convenient to be able to delete the current recording. 

At the moment you need to remember te recorded timestamp then close the dialog and lookup the recording again

I'm using Version 4.2.2 in the docker image on a Rock64 SBC with Ubuntu bionic

zagrim, 2020-03-01:
Also requested in #1425 and several others.

petrbrown, 2020-03-02:
---------- Forwarded message ----------

model (#1
the, the 657)
To: "ccrisan/motioneye" <motioneye@noreply.github.com>
Cc: "Subscribed" <subscribed@noreply.github.com>

closed 2021-03-05 by starbasessd