motioneye-project/motioneye #1659: Google drive upload error

Hernanvb, 2020-03-03:
Started getting this error after a few weeks of it working fine.
Restarting the process didn't fix the issue. Power cycling the rPi didn't fix it either.
Getting a new authorization key and creating a new folder fixes the issue temporarily but the error returns after restarting the process.
Using motioneye addon for Hassio v0.70
https://github.com/hassio-addons/addon-motioneye/tree/v0.7.0

   ERROR: failed to upload file "/share/motioneye/Camera2/2020-03-03/12-27-21.jpg" with service gdrive: 'NoneType' object has no attribute 'endswith'
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
    service.upload_file(target_dir, filename, camera_name)
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
    self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 357, in upload_data
    'parents': [{'id': self._get_folder_id(path)}]
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 394, in _get_folder_id
    if not location.endswith('/'):
AttributeError: 'NoneType' object has no attribute 'endswith'
   ERROR: failed to upload file "/share/motioneye/Camera2/2020-03-03/12-27-21.mp4" with service gdrive: 'NoneType' object has no attribute 'endswith'
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
    service.upload_file(target_dir, filename, camera_name)
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
    self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 357, in upload_data
    'parents': [{'id': self._get_folder_id(path)}]
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 394, in _get_folder_id
    if not location.endswith('/'):
AttributeError: 'NoneType' object has no attribute 'endswith'

refex, 2020-09-22:
same here, also the new key does not work even without restart. Did you manage to solve it @Hernanvb ?

refex, 2020-09-25:
I have to correct myself: it appears it fails only after a restart.

closed 2021-03-05 by starbasessd