motioneye-project/motioneye #1660: inline help question marks

jurgenweber, 2020-03-08:
I am using the docker image on a pi3 (not the motioneyeos).

I see there are '?' symbols next to options, but no text ever appears. I feel I am missing a lot and could be really helped by this inline help. Anything I can do to debug and get it working?

zagrim, 2020-03-08:
The ? icons should show helpful description when you hover that icon with mouse pointer. Are you using a touch screen and no mouse, perhaps?

jurgenweber, 2020-03-09:
no, standard MacOS, Catalina, 10.15.3, using Google Chrome.

I assumed it would be helpful description or something of the light and yes, I am not getting it. Its kinda hard to take a photo of it or something but I get nothing. I tried inspecting and checking logs, Ic an see nothing of interest.

zagrim, 2020-03-09:
Unless there's something funky with MacOS and Chrome, that could seemingly be some sort of cache issue, in which case please try clearing your cache (SHIFT + F5 / CTRL + F5). If that doesn't help, do you know having installed any extensions which might mess with handling `title` attributes in HTML tags?

jurgenweber, 2020-03-09:
I opened up the develop console and did a hard cache reload and I have htem now.

Thanks

closed 2020-03-09 by jurgenweber