motioneye-project/motioneye #1661: Added wrong url in webhook

Keldandorin, 2020-03-08:
Added following url in still picture webhook
**http://node-red.domain.com/motioneye?path=\\10.1.15.172\sdcard\Camera1\%Y-%m-%d\%H-%M-%S.jpg;Name=%Y-%m-%d\%H-%M-%S.jpg**

Guess its the **;** before Name that made it happen.

Made motioneye to say I have not added any cameras anymore

ccrisan, 2020-03-08:
Unfortunately the `;` is going to cause problems, no matter what we do about it. It's the commands separator. I recommend using the URL-encoded value `%3B` instead.

closed 2020-03-08 by ccrisan

zagrim, 2020-03-09:
@Keldandorin In case you hadn't done that already, you can get the missing cameras back by manually editing `motioneye.conf` and finding that setting and removing the part starting with the semicolon `;`.