motioneye-project/motioneye #1665: Motioneye Server use GPU HW Acceleration

nekonoire, 2020-03-10:
Hello! Been using motioneye for over a year now, and it has been one of the easiest camera software around to use. 

Recently I discovered that by using the HEVC codec option in the MotionEye web UI, I can save some storage space as the saved files are smaller. However, in doing so my CPU usage spikes up to 90% to 100% consistency. H264 option uses less CPU but takes up more space. 

My setup have motioneye running on a dedicated Ubuntu 18.04 LTS VM inside ESXI 6.7. ESXI host CPU is dual Xeon E5 2650L. VM is provisioned with 8 vCPU and 8GB RAM. I have 5 raspberry pi cameras running MotionEyeOS on them, all running in Fast Camera mode. They are set to record continuously in 1800 seconds intervals, for up to 1 month @ 15 FPS 1440 x 1024. The Ubuntu VM takes the feed from all 5 pis and saves it to a mounted NAS location. 

When set to H264, the Ubuntu VM shows around 60% CPU usage average. When set to HEVC, it goes up to 100%. I have added a Nvidia Quadro P400 pass through to the VM, rebuild the VM from scratch, installed latest Nvidia driver from Nvidia, compiled FFmpeg with the nvenc codecs, and reinstalled motioneye. CPU usage is still high, and nvidia-smi shows no GPU usage. 

Does motioneye have a way to utilize a GPU to offload the recording?

HawtDogFlvrWtr, 2020-04-17:
I just put in a pull request for this.

martikainen87, 2020-05-07:
> I just put in a pull request for this.

Nice job! Saw that it was merged, care to explain how to get it working? I tried adding:

> --gpus="all"

To the docker run command but can't see any load on the gpu running nvidia-smi. 

(also running ubuntu + docker)

HawtDogFlvrWtr, 2020-05-07:
In fairness, I've not used this within a docker container but I know you'll need to make sure the install of ffmpeg on the system supports the correct codecs. If it doesn't, it will need to be compiled. You can see if ffmpeg supports the nvidia codecs by running "ffmpeg -encoders" and checking for the right codec "nvenc".

martikainen87, 2020-05-07:
> In fairness, I've not used this within a docker container but I know you'll need to make sure the install of ffmpeg on the system supports the correct codecs. If it doesn't, it will need to be [...]

Aah, guessing that needs to be done in the docker container by ccrisan then

HawtDogFlvrWtr, 2020-05-07:
I believe you're right.

D34DC3N73R, 2020-06-30:
Just wanted to follow up on this. I'm also running motioneye in a docker container on ubuntu 18.04 and have my GPU passed into the container. However, I have same issue, ffmpeg is built without nvdec/nvenc. I've tried both master-amd64 and dev-amd64 tags. Is there a chance it could be included in a future release?

kleini, 2020-06-30:
Try to provide a merge request to get support for GPUs into the container image.

closed 2021-03-05 by starbasessd