motioneye-project/motioneye #1666: Motioneye Reboot Problem...

Bastlerstube, 2020-03-11:
Hello,

i have managed to install Motioneye according to the instructions

https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

it runs on Raspbian Buster, but after the first reboot Motioneye cannot be started anymore, neither manually nor in autostart, what am i doing wrong ?

thx

welbo97, 2020-03-24:
motion and motioneye running in an LXC Ubuntu 18.10 container. It's been working pretty well for a while. Today, while viewing one of the camera's saved videos, the web page froze. After logging into the terminal of the LXC, I found an ffmpeg process trying to display that video so decided to just reboot the LXC. After the LXC came back up, motioneye was not running. 

`
 systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-03-24 20:51:32 UTC; 6s ago
  Process: 1309 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 1309 (code=exited, status=1/FAILURE)

Mar 24 20:51:32 sonarr meyectl[1309]:     _get_additional_config(main_config)
Mar 24 20:51:32 sonarr meyectl[1309]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 2020, in _get_additional_config
Mar 24 20:51:32 sonarr meyectl[1309]:     (sections, configs) = get_additional_structure(camera=bool(camera_id))
Mar 24 20:51:32 sonarr meyectl[1309]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1991, in get_additional_structure
Mar 24 20:51:32 sonarr meyectl[1309]:     result = func()
Mar 24 20:51:32 sonarr meyectl[1309]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/tzctl.py", line 126, in timeZone
Mar 24 20:51:32 sonarr meyectl[1309]:     import pytz
Mar 24 20:51:32 sonarr meyectl[1309]: ImportError: No module named pytz
Mar 24 20:51:32 sonarr systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Mar 24 20:51:32 sonarr systemd[1]: motioneye.service: Failed with result 'exit-code'.`

motion appears to be ok and ready if called upon:

`systemctl status motion
● motion.service - LSB: Start Motion detection
   Loaded: loaded (/etc/init.d/motion; generated)
   Active: active (exited) since Tue 2020-03-24 20:42:22 UTC; 9min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 162 ExecStart=/etc/init.d/motion start (code=exited, status=0/SUCCESS)

Mar 24 20:42:22 sonarr systemd[1]: Starting LSB: Start Motion detection...
Mar 24 20:42:22 sonarr motion[162]:  * Not starting motion daemon, disabled via /etc/default/motion
Mar 24 20:42:22 sonarr systemd[1]: Started LSB: Start Motion detection.`

The error seems to point to `no module named pytz`.  I went ahead and did another `pip install motioneye -U` and it was already at 0.42. Should I try to manually install the pytz module?

@Bastlerstube What is the output of your `systemctl status motioneye` ? Is it the same error?

Bastlerstube, 2020-03-26:
systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-03-10 17:16:38 CET; 2 weeks 2 days ago
  Process: 734 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited,
 Main PID: 734 (code=exited, status=1/FAILURE)

Mär 10 17:16:34 raspberrypi meyectl[734]:     server.listen(port, address)
Mär 10 17:16:34 raspberrypi meyectl[734]:   File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver
Mär 10 17:16:34 raspberrypi meyectl[734]:     sockets = bind_sockets(port, address=address)
Mär 10 17:16:34 raspberrypi meyectl[734]:   File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.p
Mär 10 17:16:34 raspberrypi meyectl[734]:     sock.bind(sockaddr)
Mär 10 17:16:34 raspberrypi meyectl[734]:   File "/usr/lib/python2.7/socket.py", line 228, in meth
Mär 10 17:16:34 raspberrypi meyectl[734]:     return getattr(self._sock,name)(*args)
Mär 10 17:16:34 raspberrypi meyectl[734]: socket.error: [Errno 98] Address already in use
Mär 10 17:16:34 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAIL
Mär 10 17:16:38 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
...skipping...
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-03-10 17:16:38 CET; 2 weeks 2 days ago
  Process: 734 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited,
 Main PID: 734 (code=exited, status=1/FAILURE)

Mär 10 17:16:34 raspberrypi meyectl[734]:     server.listen(port, address)
Mär 10 17:16:34 raspberrypi meyectl[734]:   File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver
Mär 10 17:16:34 raspberrypi meyectl[734]:     sockets = bind_sockets(port, address=address)
Mär 10 17:16:34 raspberrypi meyectl[734]:   File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.p
Mär 10 17:16:34 raspberrypi meyectl[734]:     sock.bind(sockaddr)
Mär 10 17:16:34 raspberrypi meyectl[734]:   File "/usr/lib/python2.7/socket.py", line 228, in meth
Mär 10 17:16:34 raspberrypi meyectl[734]:     return getattr(self._sock,name)(*args)
Mär 10 17:16:34 raspberrypi meyectl[734]: socket.error: [Errno 98] Address already in use
Mär 10 17:16:34 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAIL
Mär 10 17:16:38 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.

-----------------------------------------------------------------------------------------------

systemctl status motion
● motion.service - LSB: Start Motion detection
   Loaded: loaded (/etc/init.d/motion; generated)
   Active: active (exited) since Tue 2020-03-10 16:51:38 CET; 2 weeks 2 days ago
     Docs: man:systemd-sysv-generator(8)
  Process: 348 ExecStart=/etc/init.d/motion start (code=exited, status=0/SUCCESS

Mär 10 16:51:38 raspberrypi systemd[1]: Starting LSB: Start Motion detection...
Mär 10 16:51:38 raspberrypi motion[348]: Not starting motion daemon, disabled vi
Mär 10 16:51:38 raspberrypi systemd[1]: Started LSB: Start Motion detection.

Bastlerstube, 2020-03-26:
I found the error, Motioneye was probably started twice and after the log output I could see that the port had problems even though it was unused. I have been using port 88 for ages and have now switched to port 89, now start Motioneye alone again, thanks for the inspiration.

welbo97, 2020-03-26:
Ah, ok we have different problems then. Glad you got yours worked out. 
I was able to solve my problem by installing the pytz module. 
Feel free to close the issue.

samigroup, 2020-04-13:
I had the same issue, mine I think I found the problem, but cant resolve yet. 
"  config directory "/data/etc" does not exist or is not writable "  
Here is the status output: sudo systemctl status motioneye

● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-04-13 15:09:34 EDT; 6min ago
  Process: 2529 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=255/EXCEPTION)
 Main PID: 2529 (code=exited, status=255/EXCEPTION)

Apr 13 15:09:33 raspberrypi systemd[1]: Started motionEye Server.
Apr 13 15:09:34 raspberrypi meyectl[2529]:     INFO: hello! this is motionEye server 0.42
Apr 13 15:09:34 raspberrypi meyectl[2529]: CRITICAL: config directory "/data/etc" does not exist or is not writable
Apr 13 15:09:34 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=255/EXCEPTION
Apr 13 15:09:34 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
~

Here is a copy of the config file: 
conf_path /data/etc
run_path /tmp
log_path /var/log
media_path /data/output
motion_binary /usr/bin/motion
log_level info
listen 0.0.0.0
port 80
mount_check_interval 300
motion_check_interval 10
cleanup_interval 3600
remote_request_timeout 10
mjpg_client_timeout 10
mjpg_client_idle_timeout 10
smb_shares true
smb_mount_root /data/media
wpa_supplicant_conf /data/etc/wpa_supplicant.conf
local_time_file /data/etc/localtime
enable_reboot true
enable_update true
smtp_timeout 60
zip_timeout 500
add_remove_cameras true
password_hook /usr/libexec/meyepasswd

The config path refers to the "conf_path /data/etc" and it should be ... "conf_path /etc/motioneye"
I cant seem able to update the conf file.

samigroup, 2020-04-13:
I forgot to mention and didn't pay attention as that _I personally corrupted the config, as I used a previous back up from a stand alone motionEye server, where my cameras are already setup_. if this what happens to anyone else, I was able to fix my issue by overriding the config file, created a back of the file on the desktop, updated the path and other configs, removed the /data/ from many places, as there were not correct ( according to my system), also needed to update the output location too. 

Here is the commands I followed. 
1- stop the service: 
sudo  systemctl stop motioneye

2- back up the config. 

sudo cp /etc/motioneye/motioneye.conf /home/pi/Desktop/motioneye_BK.conf
sudo cp /etc/motioneye/motioneye.conf /home/pi/Desktop/motioneye.conf

3. Open the desktop version of the config file in text editor
conf_path /etc/motioneye
run_path /tmp
log_path /var/log
media_path /tmp/MotionEye/output
motion_binary /usr/bin/motion
log_level info
listen 0.0.0.0
port 80
mount_check_interval 300
motion_check_interval 10
cleanup_interval 3600
remote_request_timeout 10
mjpg_client_timeout 10
mjpg_client_idle_timeout 10
smb_shares true
smb_mount_root /data/media
wpa_supplicant_conf /etc/motioneye/wpa_supplicant.conf
local_time_file /etc/localtime
enable_reboot true
enable_update true
smtp_timeout 60
zip_timeout 500
add_remove_cameras true
password_hook /usr/libexec/meyepasswd

4.Copy the modified config to override the system registry one.  
sudo cp /home/pi/Desktop/motioneye.conf /etc/motioneye/

5. Now you have to restart the system: 
sudo  systemctl restart motioneye
sudo  systemctl status motioneye

pi@raspberrypi:~ $ sudo  systemctl restart motioneye
pi@raspberrypi:~ $ sudo  systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-04-13 15:47:18 EDT; 2s ago
 Main PID: 7655 (meyectl)
    Tasks: 3 (limit: 2200)
   Memory: 34.5M
   CGroup: /system.slice/motioneye.service
           ├─7655 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
           ├─7664 /bin/sh -c /usr/bin/motion -h || true
           └─7665 /usr/bin/motion -h

Apr 13 15:47:18 raspberrypi systemd[1]: Started motionEye Server.
Apr 13 15:47:19 raspberrypi meyectl[7655]:     INFO: hello! this is motionEye server 0.42
pi@raspberrypi:~ $ 

Now, this is all good, I got stuck after that with the 

● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-04-13 15:47:23 EDT; 11min ago
  Process: 7655 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 7655 (code=exited, status=1/FAILURE)

Apr 13 15:47:23 raspberrypi meyectl[7655]:     _get_additional_config(main_config)
Apr 13 15:47:23 raspberrypi meyectl[7655]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 2020, in _get_additional_config
Apr 13 15:47:23 raspberrypi meyectl[7655]:     (sections, configs) = get_additional_structure(camera=bool(camera_id))
Apr 13 15:47:23 raspberrypi meyectl[7655]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1991, in get_additional_structure
Apr 13 15:47:23 raspberrypi meyectl[7655]:     result = func()
Apr 13 15:47:23 raspberrypi meyectl[7655]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/tzctl.py", line 126, in timeZone
Apr 13 15:47:23 raspberrypi meyectl[7655]:     import pytz
Apr 13 15:47:23 raspberrypi meyectl[7655]: ImportError: No module named pytz
Apr 13 15:47:23 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Apr 13 15:47:23 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
~
~
~
@welbo97, thank you!!! 
sudo pip install pytz 

Also noticed that my config was corrupted again, so updated it as it shows above. 

In my new config, the port is 80, 
so I dont need :7865 in the address bar. 

My issue is closed, too. Thanks.

closed 2021-03-05 by starbasessd