motioneye-project/motioneye #1670: SV3C Model SV-B06W-1080P-HX-2 Not Working

lacojim, 2020-03-15:
Hi. I just bought a few of these outstanding cameras on Amazon. They support RTSP and ONVIF and supposedly RTMP as well. I am using the URL: rtsp://admin:admin@192.168.x.x:554/11 to access the first stream. While this URL does work great in VLC, I cannot get it to work at all in MotionEye. I only get the gray box. 

This is the output I get when running: ffmpeg -i rtsp://admin:admin@192.168.x.x:554/11
Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080, 30 tbr, 90k tbn, 180k tbc
Stream #0:1: Audio: pcm_alaw, 8000 Hz, mono, s16, 64 kb/s

Do you have any ideas here? Is there any other information I may provide?
Thanks.

lacojim, 2020-03-16:
These are the log errors: Strange it says "No route to host" 

[1:ml1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera((null)): No route to host
[1:ml1] [ERR] [VID] vid_start: Netcam RTSP failed to open
[1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
[6:nc6:Camera3] [ERR] [NET] netcam_rtsp_handler_reconnect: Normal resolution: Reconnecting with camera....
[1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera((null)): No route to host
[1:ml1] [ERR] [VID] vid_start: Netcam RTSP failed to open
   ERROR: mjpg client timed out receiving data for camera 4 on port 8084

zagrim, 2020-03-17:
I guess you have run the tests you made with VLC and ffmpeg on a different system than you are running MotionEye? In that case you would need to fix network routing between MotionEye host and the camera.

lacojim, 2020-03-17:
All tests (VLC and ffmpeg) were done on the same system motioneye is running on.

zagrim, 2020-03-17:
In that case I'm out of guesses as it shouldn't be possible for one application to be able to connect and another unable to connect, unless the failing application is ran in a misconfigured container, or it's a weird bug in the failing application (in this case Motion).
To make it clear, are you running dockerized MotionEye?

lacojim, 2020-03-18:
No, it is running directly as a service on a Fedora 30 64-bit desktop computer. 

These are really awesome and cheap waterproof WiFi cameras. Shame I cannot get them to work in ME. So far, ME has worked perfectly on my 4 indoor cams. All different makes. These are the first I have had problems with. They also work fine in Roku camera viewer (on a smart TV) and in Kodi (on this computer) as a .strm file.

lacojim, 2020-03-20:
Well I made some very good progress. I changed the motioneye.conf **listen** address from 0.0.0.0 to the same IP of the server (192.168.1.15). I was then able to add one of my new outdoor cameras as my 4th camera. Yay!

Now the issue is, if I add anymore than four total cameras, I get random duplicates. Upon adding the other outdoor camera as Camera 5,  it showed me the video from indoor Camera 2. I deleted it, re-added it, now it shows me the video for indoor Camera 3. So I left it and added a 6th camera. It duplicates indoor Camera1. Next, I deleted the newly added cameras and added the other one. It works in the 4th camera position just fine. So both cameras work in the 4th position, but anything above four total cameras only duplicates existing cameras. How could this happen? The IPs are certainly different.

closed 2020-03-23 by lacojim