motioneye-project/motioneye #1671: Allow for an authenticated connection to the control port

hessijames79, 2020-03-15:
For my setup, I need the possibility to start and stop recording through external events (e. g. PIR sensors). The motion api offers several commands to achieve this, e. g. detection/start / detection/pause.

Since I cannot connect from localhost, I have to set webcontrol_localhost to off thus leaving the motion control api open for everyone in my local network. Hence, motionEye should allow for an authenticated connection (webcontrol_auth_method, webcontrol_authentication) to motion.

ViViDboarder, 2020-04-01:
I'm also hoping to do something similar.

It would be great to do this through a motionEye API actually, however it doesn't appear that the API is implemented yet. Is there any progress on that? I'm happy to help build it out but don't want to start from scratch if progress has already been made.

hessijames79, 2021-03-05:
@starbasessd Why exactly was this closed?

starbasessd, 2021-03-05:
The owner of the project has indicated there won't be an API developed.
There isn't any plans for https. There are instructions for running behind nginx
Last activity in this was April 1 2020 and I am trying to clean up open issues.

hessijames79, 2021-03-05:
This issue is issue is neither about https nor about a new api. Nginx won‘t solve it. I‘m afraid, blindly closing all open issues without any comment won‘t help anyone and is rather unconventional even for orphaned projects.

starbasessd, 2021-03-05:
The owner of the project has indicated there won't be an API developed.
True enough as to https. 
Since there won't be an API, my next question is what port are you connecting to?
7999?

starbasessd, 2021-03-05:
This project is not orphaned. It is in active development / support.

hessijames79, 2021-03-05:
> The owner of the project has indicated there won't be an API developed.
> True enough as to https.

> This issue is neither about https nor about a new api. Nginx won‘t solve it.

> Since there won't be an API, my next question is what port are you connecting to?
> 7999?

> Hence, motionEye should allow for an authenticated connection (webcontrol_auth_method, webcontrol_authentication) to motion.

starbasessd, 2021-03-05:
What port are you connecting to currently that you want authentication added to?

hessijames79, 2021-03-05:
> What port are you connecting to currently that you want authentication added to?

> Hence, motionEye should allow for an authenticated connection (webcontrol_auth_method, webcontrol_authentication) to motion.

In other words: The port ist completely irrelevant because it can be set to whatever you like: https://motion-project.github.io/motion_config.html#webcontrol_port

starbasessd, 2021-03-05:
OK, you are talking to port 7999 (or equivalent) which is controlled by motion, not motionEye currently. 
Add your wanted functions to /etc/motioneye/motion.conf
You can't change everything in the motion.conf file in motionEye, anyway...

hessijames79, 2021-03-05:
> OK, you are talking to port 7999 (or equivalent) which is controlled by motion, not motionEye currently.
> Add your wanted functions to /etc/motioneye/motion.conf
> You can't change everything in the motion.conf file in motionEye, anyway...

I can do that but motionEye can't connect to it any more. That is what this issue is about.

starbasessd, 2021-03-06:
I just took a motion/motionEye on Ubuntu 20.04 server
Added 
thread thread-1.conf
webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost off
webcontrol_interface 1
webcontrol_auth_method 1
webcontrol_authentication admin:password
 to /etc/motioneye/motion.conf
Connected to motioneye:8765 and got all 5 cameras.
Connected to motioneye:7999, was prompted for username & password, and was let in.
Using 

motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Ubuntu 20.04
What kind of issue are you having?

hessijames79, 2021-03-06:
> Connected to motioneye:8765 and got all 5 cameras.
Of course you did. The streams are acquired through stream_port, not webcontrol_port.

> Connected to motioneye:7999, was prompted for username & password, and was let in.
Of course you did. You used solely motion (not motionEye) functionality.

> What kind of issue are you having?
My issue is that if I enable webcontrol-authentication motionEye can't connect to the webcontrol port any more. So everything will work fine till you use motionEye functionality that needs the webcontrol connection, e. g. pressing the snapshot button:
   ERROR: failed to take snapshot for camera with id 5: authentication failed

starbasessd, 2021-03-06:
Mar  5 19:34:34 camera-server meyectl[614]:    ERROR: failed to take snapshot for camera with id 5: authentication failed
Thanks for being so helpful there, @hessijames79 I now have something to point people at.

hessijames79, 2021-03-06:
> Mar 5 19:34:34 camera-server meyectl[614]: ERROR: failed to take snapshot for camera with id 5: authentication failed
> Thanks for being so helpful there, @hessijames79 I now have something to point people at.

Thank you. But please let me point out that all necessary information was already in the very first post.

starbasessd, 2021-03-06:
Took me quite a while to reproduce it without assistance from you. First
step in checking for a bug or issue is to reproduce it, Posting the
specific error, with the specific versions of software (apps and OS) go a
long way to help get it resolved/

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-06:
Outside the box:
Have you thought about an ssh tunnel?
That way you can leave webcontrol_localhost on
and only people with ssh access would be able to use the motion API. 
Since you are using motion/motionEye on an OS, you have the full capabilities of SSH ACLs, etc...

hessijames79, 2021-03-06:
> Have you thought about an ssh tunnel?
I have thought about various workarounds different tunnel implementations being just one of them. In fact, I have been using one since creating this video. But no workaround beats a clean and working implementation.

closed 2026-10-25 by MichaIng