motioneye-project/motioneye #1675: Unable to add rtsp camera - worked previously

Pi-DD, 2020-03-17:
In the process of migrating my MotionEye server to a Buster machine (from Stretch). This is a completely new config on a Pi4b, not an upgrade in place.

The existing server runs Stretch 9.11, MotionEye 0.39.3, Motion 4.1.1
New server runs Buster 10.0, MotionEye 0.42, Motion 4.2.2

On the existing server I have a number of Wifi cameras (Tenvis models) - all configured and working. 
When I add the ones that stream via http, to the new server they work.
When I try to add the rtsp ones - nothing - I just keep getting "connection closed". Once, briefly I saw a camera model flicker past, in the lower config box, but nothing since. 

I've tried putting the user/pw before and after the URL, which I don't remember doing before, still nothing.  The URL I am using works just fine to VLC on both iOS and other Pi's.

The cameras support Onvif, which I have turned on, but the older server worked fine without that.

I've tried rtmp - it finds the camera but all that appears is a blank grey screen with "Unable to open video device since   timexxxxx".

Anyone got a clue how to fix this? Thanks in advance.

steinhorstbr, 2020-04-10:
I have this problem also in motioneye python 3 docker

Pi-DD, 2020-04-10:
My “fix” was to reboot the camera! For some reason this fixed the problem.

closed 2021-03-05 by starbasessd