motioneye-project/motioneye #1678: Docker is not building

FredClement91, 2020-03-22:
If that can help improving the product, I get following error when building the docker :

> Step 9/16 : RUN pip install motioneye==$MOTIONEYE_VERSION
>  ---> Running in 12c4b5fd110f
> Downloading/unpacking motioneye==0.37
>   Running setup.py (path:/tmp/pip_build_root/motioneye/setup.py) egg_info for package motioneye
>     
> Downloading/unpacking tornado>=3.1 (from motioneye==0.37)
>   Running setup.py (path:/tmp/pip_build_root/tornado/setup.py) egg_info for package tornado
>     /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
>       warnings.warn(msg)
>     
>     warning: no files found matching '*.yaml' under directory 'demos'
>     no previously-included directories found matching 'docs/build'
>     warning: no files found matching 'tornado/test/README'
> Downloading/unpacking jinja2 (from motioneye==0.37)
> Downloading/unpacking pillow (from motioneye==0.37)
>   Running setup.py (path:/tmp/pip_build_root/pillow/setup.py) egg_info for package pillow
>     Traceback (most recent call last):
>       File "<string>", line 17, in <module>
>       File "/tmp/pip_build_root/pillow/setup.py", line 304
>         yield from self.features
>                  ^
>     SyntaxError: invalid syntax
>     Complete output from command python setup.py egg_info:
>     Traceback (most recent call last):
> 
>   File "<string>", line 17, in <module>
> 
>   File "/tmp/pip_build_root/pillow/setup.py", line 304
> 
>     yield from self.features
> 
>              ^
> 
> SyntaxError: invalid syntax
> 

Sadly, I am not an python expert at all and I don't know what it means.

kleini, 2020-03-23:
motioneye 0.37??? This is completely outdated. We're today now at version 0.42. You should update your cloned source code repositories.

closed 2020-03-23 by kleini

FredClement91, 2020-03-23:
Thanks for your reply,

Sorry, I will do that indeed.

Fred