motioneye-project/motioneye #1680: alprd is consumming cpu

acognard, 2020-03-27:
hello, 
using debian 10, alprd is really consumming cpu
I stopped server
 meyectl stopserver -b -c /etc/motioneye/motioneye.conf
but process is still running
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
 2240 root      20   0  770172  71688  35456 S 205,9   2,3   1604:08 alprd
 1354 www-data  20   0  239288  34216  24516 R  29,4   1,1  43:07.69 php

stopping motion
/etc/init.d/motion stop

same result

Any help ?

akidburn, 2020-03-27:
@acognard 
uninstall alprd, as fast as i know its not used by motion or motioneye

acognard, 2020-03-27:
well, this is curious, not able de find the corresponding paquage :(

# ps -u root | grep alprd
 1909 ?        00:00:08 alprd
 1916 ?        00:47:36 alprd
 1917 ?        00:08:20 alprd

off course, # apt-get purge alprd does not work

CRImier, 2020-03-27:
Try this to know which package installed the file:
````
dpkg -S `which alprd`
````

acognard, 2020-03-27:
# dpkg -S `which alprd`
dpkg-query: aucun chemin ne correspond à /usr/bin/alprd

CRImier, 2020-03-27:
The backticks were required, so the first command executed correctly. Any chance you installed this manually? Try using `stat /usr/bin/alprd` to see the last file modification date - perhaps, you can remember yourself tinkering with your system at that date.

acognard, 2020-03-27:
thank you very much to all for your help !!!
not installed using motioneye, but jeedom  openalpr plugin !!! 

have a nice week end, and take care :)

closed 2020-03-27 by acognard

CRImier, 2020-03-27:
happy to help, take care!