motioneye-project/motioneye #1681: Is there a way to install and run motioneye on MacOS ?

danch99, 2020-03-28:
Hi,

Just wondering, I have a Mac mini as a server and several pi zero with cameras and motioneyeOS, I’d like to install motioneye on the mac to use it as a motioneye hub.
Is there a way to install motion eye on Mac OS as it’s linux based ?

Thanks

shaynethiessen, 2020-04-08:
You could use the docker image.

runestate, 2020-07-15:
Does not work for me: If I run the docker-compose file, access the UI (http://localhost:8765/) and try to add a running pi zero/motioneye, then i get an error "an error occurred".  From the logs it looks like the remote zero. itself is requesting video config from the local (mac) machine, which then fail's with "Failed to open /dev/video0: No such file or directory".

closed 2021-03-05 by starbasessd