motioneye-project/motioneye #1685: USB Webcam not showing video in webgui

gexk0, 2020-04-03:
Hi, I want to add and an old webcam to motioneyeos. I'm using rpi 1 model B. the command
  v4l2-ctl --list-devices shows as follows:
[root@yarn ~]#  v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12

CIF Single Chip      (usb-20980000.usb-1.2):
        /dev/video0

But  it is unable to show device when I add it to webgui

thelostinthought, 2020-06-28:
same problem with a pi 3b+ and a logitech c920

thelostinthought, 2020-06-30:
same problem with a pi 3b+ and a logitech c920 in docker with IOTstack.  turns out that IOTstack comments out the last 2 lines of ~/IOTstack/docker-compose.yml.  it should read:

    devices:
      - "/dev/video0:/dev/video0"

uncommenting them and restarting solved my issue.

closed 2021-03-05 by starbasessd