motioneye-project/motioneye #1690: run without installation as service

vkuehn, 2020-04-06:
after git clone the repo is there a way to run motioneye from command line with having the whole setup or just a venv installation ?

HJSK, 2020-04-07:
Whats a venv installation? 
I also used the repo and run in several problems, files like /usr/local/share/motioneye/extra/motioneye.systemd-unit-local are not there. Even total directory from /usr/local/share/ are missing. Same for motioneye.conf.sample, not to find in the repro / installation procedure.
After looking here on github I found the file but think is not the right one or has something to do with rights. 
I want to give this repo way a try while in motioneyeOS I run into a overheating from CPU (higher then 80 degrees!) with use of 2-3 IP rtsp CAM's

ccrisan, 2020-04-12:
> after git clone the repo is there a way to run motioneye from command line with having the whole setup or just a venv installation ?

You can  run `python2 motioneye/meyectl.py startserver` or `python2 -m motioneye.meyectl startserver`. You may have to adjust your `PYTHONPATH` though.

vkuehn, 2020-04-13:
there is no python3 support ?

vkuehn, 2020-04-13:
@HJSK Venv as in python3 -m vena
@ccrisan with that I get
 Traceback (most recent call last):
  File "motioneye/meyectl.py", line 275, in <module>
    main()
  File "motioneye/meyectl.py", line 248, in main
    load_settings()
  File "motioneye/meyectl.py", line 57, in load_settings
    for i in xrange(1, len(sys.argv)):
NameError: name 'xrange' is not defined

ccrisan, 2020-04-15:
@vkuehn there's no stable Python3 support yet. There's an [experimental branch](https://github.com/ccrisan/motioneye/commits/python3) though that you can use with Python 3.

closed 2020-04-15 by ccrisan