motioneye-project/motioneye #1691: cannot add MJPEG camera (already added before)

sorriso93, 2020-04-08:
Hello
strange problem.
I used to have 7 cameras (all via motieneye remote except 2 mjpeg) and it was all ok, then I removed the 7th one week ago by mistake.
Now I can't add it anymore: i get always the error "An error occurred. Refreshing is recommended.". 
I deleted the cache of the browser and trie with Chrome and Safari, no way same error.
I checked on /data/etc and I have 7 conf files, the 7th is 0 byte. Removed the file, but in any case same error.
Suggestions?
Many thanks

zagrim, 2020-04-09:
Please check also `/data/etc/motion.conf` to see if there is still a reference to the 7th camera config file, and if so remove it.

sorriso93, 2020-04-11:
thks

sorriso93, 2020-04-13:
removed both references but same problem... is there another place where I can delete reference?

zagrim, 2020-04-14:
I think you should attach some logs here. Maybe the error that occurs is not about having a lingering file or clashing camera IDs but something else.

sorriso93, 2020-04-14:
Problem solved, I did not have additional space on sd card (100% full of log)

closed 2020-04-14 by sorriso93