motioneye-project/motioneye #1693: gpg/gpg encryption for emails

klausmcm, 2020-04-09:
Is encrypting emails with gpg planned? When sending potentially sensitive images, I would like them to be encrypted. I know there is an option to run a command under **Motion Notifications**. Would making the name of the generated image/video as a variable be an options?

I'm happy to share the function for creating gpg encrypted ciphertext of an email with attachments if that helps.

BeWog, 2020-04-18:
You can use this option [on_movie_end](https://motion-project.github.io/motion_config.html#on_movie_end) and use your own script to send encrypted emails.

klausmcm, 2020-04-26:
> You can use this option [on_movie_end](https://motion-project.github.io/motion_config.html#on_movie_end) and use your own script to send encrypted emails.

Thank you, I will check that out.

closed 2021-03-05 by starbasessd