motioneye-project/motioneye #1694: Firefox 75/Chrome 81 cannot view cameras

STrRedWolf, 2020-04-09:
**Reproduction steps:**

1. Set up MotionEye on a Ubuntu server (LTS or latest, doesn't matter).
2. Attach one or more MotionEyeOS cameras (I used Raspberry Pi Zero W's)
3. Go to the server (port 8765) from Firefox 75 or Chrome 81

**Expected Results:** Cameras show.

**Actual Results:** Camera loading icons keep spinning.

**Notes:** Connecting directly to the cameras on separate tabs works (see camera input), so nothing wrong with MotionEyeOS.  Previous versions (Firefox 74.0.x and Chrome 80) also work properly.  

Versions used:

- Server:  MotionEye 0.39.3; Motion 4.2.2; Ubuntu 20.04
- Cameras:  Raspberry Pi Zero W's, MotionEyeOS 20200203 and 20180224

nx-99, 2020-04-09:
Same problem with Operating System: Ubuntu 18.04.4 LTS and Firefox 75 and Google Chrome
Versione 81.0.4044.92 (Build ufficiale) (a 64 bit).
Regard.

STrRedWolf, 2020-04-09:
Let me add one thing here.  The OS of the server or cameras does not matter.  The OS the web browsers does not matter (I can reproduce on Linux and Windows).  The version of the web browser matters.

nx-99, 2020-04-09:
motionEye Version	0.39.1, Motion Version	4.1.
With falkon Webengine browser 3.0.0 QtWebEngine 5.9.5 work.
I think the problem is firefox 75.
Regard.

STrRedWolf, 2020-04-09:
Found a problem.  The "loading" attribute on IMG tags has been implemented in Firefox 75 and Chrome 81, per their release notes.  It can ether be "eager" or "lazy".  MotionEye uses this attribute in a different (non-W3C spec) way and it's reading "NaN".

ccrisan, 2020-04-09:
This issue has been fixed in https://github.com/ccrisan/motioneye/commit/b22ef62dfa02dc5b09c84fca557f561043eb6f78.

Please upgrade to 0.41 or 0.42.

closed 2020-04-09 by ccrisan