motioneye-project/motioneye #1696: Python2.7 EOL'ed, Ubuntu 20.04 deprecates it

STrRedWolf, 2020-04-09:
It took a while, but I was able to upgrade to 0.42... but in my previous efforts to resolve the issue, I ended up upgrading to pre-release beta Ubuntu 20.04.

It no longer carries python2 by default, and removes python2's pip.  [Phoronix article about the removal.](https://www.phoronix.com/scan.php?page=news_item&px=Python-2-EOL-Ubuntu-20.04)

Here's what I got when I was able to upgrade it after manually installing pip.

```
root@sidney:~# pip install motioneye --upgrade
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting motioneye
  Using cached motioneye-0.42.tar.gz (757 kB)
Requirement already satisfied, skipping upgrade: tornado<6,>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.1)
Requirement already satisfied, skipping upgrade: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye) (2.10)
Requirement already satisfied, skipping upgrade: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.2.0)
Requirement already satisfied, skipping upgrade: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye) (7.43.0.2)
Requirement already satisfied, skipping upgrade: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.4.0.3)
Requirement already satisfied, skipping upgrade: futures in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.2.0)
Requirement already satisfied, skipping upgrade: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (0.5)
Requirement already satisfied, skipping upgrade: MarkupSafe>=0.23 in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye) (1.0)
Requirement already satisfied, skipping upgrade: six in /usr/local/lib/python2.7/dist-packages (from singledispatch->tornado<6,>=3.1->motioneye) (1.14.0)
Building wheels for collected packages: motioneye
  Building wheel for motioneye (setup.py) ... done
  Created wheel for motioneye: filename=motioneye-0.42-py2-none-any.whl size=790366 sha256=2993ec382a0652caf075ef0d34d461ff8b436b2fb1e3c65529482a71934c4bd7
  Stored in directory: /root/.cache/pip/wheels/89/5a/74/d7e60f930c14b53b385d64b69080e1637a75f63c2c2cad2e72
Successfully built motioneye
Installing collected packages: motioneye
  Attempting uninstall: motioneye
    Found existing installation: motioneye 0.39.3
    Uninstalling motioneye-0.39.3:
      Successfully uninstalled motioneye-0.39.3
Successfully installed motioneye-0.42
root@sidney:~# 
```

poundy, 2021-02-10:
Situation is getting more dire I'd suggest.... 

DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.

starbasessd, 2021-02-10:
I created the instructions for Ubuntu, 20.04 and newer, because of these upstream issues. It also works in other debian based flavors, at least...

closed 2021-02-10 by starbasessd

poundy, 2021-02-10:
> I created the instructions for Ubuntu, 20.04 and newer, because of these upstream issues. It also works in other debian based flavors, at least...

do you mean this wiki page? https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29

If so, then it still refers to Python2, and it still reports those errors... I know that because I followed that page, and got the errors reported.

starbasessd, 2021-02-10:
True, but until the programmers migrate away from 2.7 which has been
deprecated since 2020, and they announced the EOL back in 2014, we who
support do the best we can...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STrRedWolf, 2021-02-10:
https://lists.ubuntu.com/archives/ubuntu-devel/2020-February/040918.html

Python 2.7 won't be in 22.04 for sure.  Looking at the message in the "python-is-python2" package:

```symlinks /usr/bin/python to the DEPRECATED python2
In Ubuntu, all python packages use explicit python3 or python2 interpreter and ▒
do not use unversioned /usr/bin/python at all. Some third-party code may still ▒
be python2 based, yet may use /usr/bin/python.                                 ▒
                                                                               ▒
This is a convenience package which ships a symlink to point /usr/bin/python   ▒
interpreter at the current default python2. It may improve compatibility with  ▒
obsolete 3rd-party software, whilst breaking some modern software.             ▒
                                                                               ▒
This package will be installed upon upgrades to Ubuntu 20.04, if DEPRECATED    ▒
python2 was installed.                                                         ▒
                                                                               ▒
python2 is DEPRECATED and will not be provided in the future Ubuntu release. It▒
is recommended to remove python2 and this package after ensuring that only     ▒
python3 is in use.                                                             ▒
                                                                               ▒
No packages may declare dependencies on this package.        
```

Yeah... time to port it to Python 3.

poundy, 2021-02-10:
> True, but until the programmers migrate away from 2.7 which has been deprecated since 2020, and they announced the EOL back in 2014, we who support do the best we can...

I guess my point is, there's no point closing this issue just because we're doing the best we can. It should be tracked - there's a need to move to Python3. (I am not a developer, and not even a linux person, just a random user reporting an error, so can't help deal with anything other than reporting the behaviour)

starbasessd, 2021-02-10:
THere are many other open issues and pull requests already tracking the
issue....

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STrRedWolf, 2021-02-10:
Reference  #1572 then.

poundy, 2021-02-10:
And https://github.com/ccrisan/motioneye/issues/1640 for discussion as well. But #1696 was the newest one I landed on to comment on...