motioneye-project/motioneye #1697: Action button issue

kmsgli, 2020-04-10:
Hey how are you, I cant seem to get my PTZ functions to work with the action buttons. Both the action button script and the python script it triggers work outside of motioneye and move my camera. But when I hit the action button nothing happens and the log is telling me this:
Apr 10 08:44:00 archdesktop meyectl[9692]:  WARNING: up_6: command has finished with non-zero exit status: 1
Apr 10 08:44:00 archdesktop meyectl[9692]:  WARNING: up_6: Traceback (most recent call last):
Apr 10 08:44:00 archdesktop meyectl[9692]:  WARNING: up_6:   File "/etc/motioneye/up.py", line 2, in <module>
Apr 10 08:44:00 archdesktop meyectl[9692]:  WARNING: up_6:     from amcrest import AmcrestCamera
Apr 10 08:44:00 archdesktop meyectl[9692]:  WARNING: up_6: ModuleNotFoundError: No module named 'amcrest'
Apr 10 08:44:03 archdesktop meyectl[9692]:  WARNING: up_6: command has finished with non-zero exit status: 1
Apr 10 08:44:03 archdesktop meyectl[9692]:  WARNING: up_6: Traceback (most recent call last):
Apr 10 08:44:03 archdesktop meyectl[9692]:  WARNING: up_6:   File "/etc/motioneye/up.py", line 2, in <module>
Apr 10 08:44:03 archdesktop meyectl[9692]:  WARNING: up_6:     from amcrest import AmcrestCamera
Apr 10 08:44:03 archdesktop meyectl[9692]:  WARNING: up_6: ModuleNotFoundError: No module named 'amcrest'

It seems python can not find the amcrest module but outside of motioneye it finds the module and moves my camera just fine. Am i missing something here like how to tell motioneye where the python path is?

Thanks for the hlep

kmsgli, 2020-04-10:
I have tried doing it the recommended way using a shell script to load python script and to load to python script directly, same outcome.

On the top of the shell script i have #!/bin/bash with no file extensions

On the top of the python script I have #!/usr/bin/python with the .py file extension.

I am am out of ideas as the script works any way I do it until its run via action button in the motioneye front end.

zagrim, 2020-04-11:
Did you originally install amcrest module as root or as an individual user? If the latter, it cannot be found as MotionEye is running as a different user. In that case, just re-install amcrest by as root (e.g. running `sudo pip install` instead of `pip install`).

kmsgli, 2020-04-11:
> Did you originally install amcrest module as root or as an individual user? If the latter, it cannot be found as MotionEye is running as a different user. In that case, just re-install amcrest by as root (e.g. running `sudo pip install` instead of `pip install`).

Thanks zagrim. Installing the Amcrest module with sudo did the trick now I have all the direction action buttons working correctly. I will close this thread.

Thanks again.

closed 2020-04-11 by kmsgli