motioneye-project/motioneye #1698: Motioneye straming not working on Firefox 75.0

glyderman8, 2020-04-10:
Hi, I have just found out today, 10-04-20, that my motioneye viewing on raspian stretch was not working on two camera I use it on, when I tried to view the stream on Firefox. Thinking something had changed I updated raspian and updated motioneye to latest version. It made no difference. I then decided to use my phone. It worked ok on both cameras. One uses a webcam and other uses the raspberrypi camera. So I then realised it was probably firefox. I then used a laptop I hadn't used for a while and firefox worked ok. It was version 74.01. I let it update to version 75 and tried again. It then did not see the stream from motion eye. So latest firefox has broken the viewing capablity from moitoneye. I noticed a workaround for chrome and tried similar with firefox. That is: about:config and change the lazy-image flag to false. Guess what... it worked. So for now this workaround works.

Rouzic6666, 2020-04-17:
Hi, I have the same issue on firefox 75.0 not working, Straming working on other PC with firefox 73.0
 lazy-image flag to false not working for me...

clnhlzmn, 2020-04-18:
@Rouzic6666 the full name of the flag is `dom.image-lazy-loading.enabled`. Setting that to false fixed this for me.

aslmx, 2020-06-01:
> @Rouzic6666 the full name of the flag is `dom.image-lazy-loading.enabled`. Setting that to false fixed this for me.

Gosh... thanks! I had searched some time how to fix this. It was really annoying that it worked fine in waterfox, but did not in firefox. I couldn't figure out the difference and would never have thought of something like this after trying to investigate this with the developer console.

**Anyway... thanks!** 

Gonna write this down somewhere and hope i can find it again when i need it. I hope this behaviour gets somehow improved (I don't know though whether it has to be firefox or motioneye) so that manual hacks in about:config are not necessary in the near future.

zagrim, 2020-06-02:
Hey guys, you can get rid of the manual hack by upgrading MotionEye 0.41 or later. See #1428.

donvipre, 2020-07-07:
@ccrisan: This can be closed as it was fixed in 0.41 or later

closed 2020-07-07 by ccrisan