motioneye-project/motioneye #1701: Can I use MotionEyeOs for Commercial product

ashanuyangoda, 2020-04-12:
Hi Ccrisan,

I am Ashan.  I am from Sri Lanka. I am an Electronic Engineer with a dream of making my own company in the future. I would like to use MotionEyeOs for commercial product (kind of a Motion detection Surveillance Camera with Alarm and Light). So I would like to know the answers for the following questions.
(NOTE: I am not suppose to develop your product in any way. I am just want to use MotionEyeOs as it is.)

1) Do I need to pay you, if i am going to use your MotionEyeOs in my commercial product? 
2) When i am going to use your MotionEyeOs in my product, what are the conditions I have to follow?
3) Do I need to state credit for you in my product user manual?

Thanks
Ashan

ccrisan, 2020-04-12:
> 1. Do I need to pay you, if i am going to use your MotionEyeOs in my commercial product?

You definitely **don't need** to pay me anything for using motionEye(OS) commercially. If you really find it helpful, a donation is always welcomed, though.

> 2. When i am going to use your MotionEyeOs in my product, what are the conditions I have to follow?

motionEye is GPLv-3, while motionEyeOS has a large amount of opensource licenses being based on [BuildRoot](https://buildroot.org/) and thousands of other opensource projects. Long story short, just provide links to the source code and don't claim it's your work, and you should be fine.

> 3. Do I need to state credit for you in my product user manual?

You could mention the fact that it runs an opensource project and provide a link to the GitHub page. You don't need to mention my name or any other authors'.

ashanuyangoda, 2020-04-12:
Hi Crisan,

Thank you very much sir for answering my questions. That is really helpful. I hope i will be able to give you a donation in the future. I will follow your guidelines.
By the way I would like to know the answers of the following questions as well,
1) Are motionEye and motionEyeOS different projects?
2) Could you please explain the difference of motionEye and motionEyeOS clearly? 
(For the moment I am using the Raspberry Pi 3 version of the motionEyeOS.)

Thanks
Ashan

ccrisan, 2020-04-15:
> 1. Are motionEye and motionEyeOS different projects?

Yes, please check out the [wiki](https://github.com/ccrisan/motioneyeos/wiki).

> 2. Could you please explain the difference of motionEye and motionEyeOS clearly?
(For the moment I am using the Raspberry Pi 3 version of the motionEyeOS.)

motionEyeOS is an OS that runs motionEye. There's no significant performance or functional difference between the two setups, it's just that the OS is ready to be used and offers integrated firmware updates, while a plain motionEye setup would require more user configuration and maintenance.

closed 2020-04-15 by ccrisan